LiberTEM/LiberTEM #1: pycroscopy

sk1p, 2018-01-16:
https://pycroscopy.github.io/pycroscopy/about.html

> pycroscopy is a python package for **image processing** and scientific **analysis** of imaging modalities such as multi-frequency scanning probe microscopy, scanning tunneling spectroscopy, x-ray diffraction microscopy, and transmission electron microscopy. pycroscopy uses a data-centric model wherein the raw data collected from the microscope, results from analysis and processing routines are all written to **standardized hierarchical data format (HDF5) files** for traceability, reproducibility, and provenance.

uellue, 2018-01-17:
About Pycroscopy I like the separation of index and payload data. 

* You can work on the index for alignment, drift correction etc. of a data set without traversing TBs of data. 
* You can map functions on the payload without worrying on spatial arrangement of the data points
* You can define arbitrary sub-selections that are just a list of pointers
* And most importantly, you can support the weirdest scan patterns or stage movements.

One issue with the format is that I couldn't wrap my head around how EXACTLY that index works. It is a bit non-intuitive to me. Maybe it would be possible to use flat lists? n frames means n x coordinates, n y coordinates, and perhaps n stage positions, n focus values, n aberrations, and so forth. THAT I could understand with my puny brain! 

The lists themselves can grow quite large and waste space, that means efficient options for default cases. Perhaps options for each parameter:

1. Constant value
2. Linear range from start to stop
3. Rectangular grid (special case for x,y pairs)
4. Smooth spline?
5. Arbitrary list

A file could perhaps have several versions of an index. As acquired, after drift correction, after cross-correlation or reconstruction, ...

CompPhysChris, 2018-08-07:
If you are interested in pycroscopy's data format, I recommend you look [here](https://pycroscopy.github.io/pyUSID/data_format.html).  We have tried to split the I/O and engineering components of the package from the analysis.  The analysis and science specific code has remained in pycroscopy, and the I/O has been moved into the new pyUSID package.  Please have a look at our documentation and feel free to ask questions or give comments in our google groups ([pycroscopy](https://groups.google.com/forum/#!forum/pycroscopy), [pyUSID](https://groups.google.com/forum/#!forum/pyusid)).