LiberTEM/LiberTEM #4: CERN ROOT

sk1p, 2018-01-16:
https://root.cern.ch/

> ROOT is ...
> A modular scientific software framework. It provides all the functionalities needed to deal with big data processing, statistical analysis, visualisation and storage. It is mainly written in C++ but integrated with other languages such as Python and R.

Interesting for our project may be [the ROOT file format](https://root.cern.ch/input-and-output) as a possible alternative to HDF5. See also [this discussion on the ROOT forums](https://root-forum.cern.ch/t/root-file-vs-hdf5/20115/2) for some comparison points.

uellue, 2018-01-17:
Hm, do we need any of these points?

uellue, 2021-02-16:
Closing here since LiberTEM took a different path. :-)

closed 2021-02-16 by uellue