LiberTEM/LiberTEM #6: MSA / MAS / AMAS HyperDimensional Data File

sk1p, 2018-01-17:
http://www.csiro.au/luminescence/HMSA/index.html

> The MSA / MAS / AMAS HyperDimensional Data File (HMSA, for short) is intended to be a common format to permit the exchange of hyper-dimensional microscopy and microanalytical data between different software applications.
> [...]
> In addition to storing hyperdimensional data, the HMSA file format is suitable for storing conventional microscopy and microanalysis data such as spectra and images, as well as experimental conditions and other metadata.

sk1p, 2018-01-17:
HMSA File Format overview: https://www.microscopy.org/resources/scientific_data/MM2011-HMSAFormat.pdf

Specification: https://www.microscopy.org/resources/scientific_data/HMSA_Specification-20161100.pdf

uellue, 2018-01-17:
With that one I have a number of issues:

* Which software packages support this?
* How usable, robust and fast is the implementation?
* The "libhmsa" development seems to be an army of one without any development process, bug tracker and such. Not good.
* The toolchain feels incomplete. Feels like Windows-only.
* The spec is outright weird
  * " subset of the XML version 1.0 format" (details on p. 17) WHY??? Excludes all XML validators!
  *  compression explicitly discouraged (p. 14) --> Not good for us, we WANT compression! 
  * Explicitly not as primary format or when performance is critical -- why use it at all, then?

In general, I'd say this is not suitable for us.

uellue, 2018-01-17:
I'll close the issue, unless someone wants to fix this mess!

closed 2018-01-17 by uellue

sk1p, 2018-01-17:
In the overview document, they write that they actually do have a schema, but I couldn't find it anywhere on their site, and the spec then specifically excludes XML Schema... :man_shrugging: - using a subset of XML in this fashion is actually not that bad, because it eliminates some weird XML security issues (billion laughs, remote entity inclusion etc.) and reduces overall complexity. You can still parse it with any regular XML parser, you just need to limit yourself to the subset on writing.

I agree on your comments about libhmsa, compression, robustness and performance.