LiberTEM/LiberTEM #7: LiberTEM should have a homepage

uellue, 2018-01-17:
For a proper communication of the project we'd need a homepage. The following items should probably be on it. What do you think, what would you add or remove?

* General description and mission
* Licensing and documentation
* Downloads
* Link to resources (mailing lists, Gitter, Github)
* Sponsors and funding (H2020: ERC Proof-of-Concept 2017, VIDEO, 780487; FZ Jülich [ER-C]; which other institutes can officially support this?)
* Contributions and acknowledgements
* Publications
* How to cite
* User stories
* Imprint and contact
* News and events, perhaps?
* Some more people-related content to give it a human face, who we are and why we are doing this?
* Development roadmap?
* Governance?

uellue, 2018-01-17:
Probably we should start with some basics and grow it over time.

uellue, 2018-01-17:
++ how to contribute

woozey, 2018-01-17:
Certainly, which platform you want to use? Something to consider, HS have their website on Github too:
https://github.com/hyperspy/hyperspy.github.com

magnunor, 2018-01-17:
It is easy to generate a webpage using (for example) Sphinx: http://www.sphinx-doc.org/en/stable/.

This is used in HyperSpy, and I use if for all of my projects. Example: http://atomap.org/ , source for that page: https://gitlab.com/atomap/atomap/blob/master/doc/index.rst (you might have to click "display source" to see the actual "code" used to generate the html.

jan-car, 2018-01-17:
:+1:  for Sphinx! Although I thought it usually is used for documentation/user guide web pages (I only used it for this up till now). Is it easy to extend to a full web presence?

sk1p, 2018-01-17:
I think Sphinx is a good start. Yes, it is meant for documentation, but you can fit a lot of content into that format. If the site gets too complex for Sphinx, I would suggest we switch the non-documentation part to a static site generator, like [pelican](https://blog.getpelican.com/), [lektor](https://www.getlektor.com/), [jekyll](https://jekyllrb.com/), or if we want to be fancy, [Gatsbyjs](https://www.gatsbyjs.org). We can keep generating the docs with Sphinx of course!

For hosting, either we can use github, or something like [netlify](https://www.netlify.com/) (some [examples of other open source project hosting their websites there](https://www.netlify.com/open-source/#mooseheads))

sk1p, 2018-01-19:
As an examlpe of a complete website built using Sphinx, see [scipy/scipy.org](https://github.com/scipy/scipy.org)

sk1p, 2018-05-30:
The documentation is now built automatically and deployed from the master branch to https://libertem.github.io/LiberTEM/ - more content coming soon!

uellue, 2018-12-13:
The homepage is pretty complete since the 0.1. We can open separate issues in case we miss specific parts.

closed 2018-12-13 by uellue