LiberTEM/LiberTEM #8: Existing off-line data analyses packages

woozey, 2018-01-17:
Here is an overview of existing packages for (S)TEM data analyses.
**Free (and open source):**
* Hyperspy #2 

**Proprietary:**
* Gatan GMS (or DM)

uellue, 2018-01-17:
What is the license for Semper? I know that we have the source code. But I am not sure if it qualifies as "open source", i.e. if we can modify and redistribute it.

woozey, 2018-01-17:
@uellue I didn't meant Semper is open source, I meant it is free... I don't think it actually has any license. I would contact Rafal about further details of Semper. There is not much to discuss about it except that it is a high level image processing language that was written in presumably 80s on Fortran. It goes back to the times when simple things like FFT was not implemented in common libraries. Nowadays it's only value is im some TEM data processing routines that were written by ex-Cambridge people including holography functions of Rafal. As far as I know there is only one person who somehow supports and maintains it.

magnunor, 2018-01-23:
- `fpd_data_processing` https://fast_pixelated_detectors.gitlab.io/fpd_data_processing/

My library for analysing pixelated STEM datasets, which builds upon HyperSpy.

uellue, 2018-01-24:
* pyFAI http://pyfai.readthedocs.io/en/latest/pyFAI.html
The do almost exactly the same operation: Integration over portions of a detector using matrix multiplication. The only difference is that they have many very sparse matrices and make an effort to be sub-pixel precise, while we have in a first approximation few fat matrices and our data is usually so blurred (large diffraction spots for small probe) that sub-pixel precision is less critical.

dnjohnstone, 2018-04-27:
As my comment in #2 (You may also like to note this: https://github.com/pyxem/pyxem which also builds on hyperspy for scan and record a diffraction pattern at every probe position type experiments.)

magnunor, 2018-06-17:
I just renamed `fpd_data_processing` to `pixStem`: http://pixstem.org/

I updated the links in this issue