LiberTEM/LiberTEM #14: Mathematical formulation to allow using math libraries for applying masks to frames

uellue, 2018-02-07:
For the mathematical operation it doesn't matter in what order or arrangement the frames are scanned. That means the "scan dimension" can be flattened to a linear array. The same way the arrangement of the pixels in a frame doesn't matter, that means masks and frames can be flattened to linear arrays as well.

With those extra dimensions flattened, applying a set of _m_ masks (_B_) to a collection of _n_ frames with _c_ pixels per frame (_A_) can be expressed as a [rectangular matrix product](https://en.wikipedia.org/wiki/Matrix_multiplication#Rectangular_matrices). _A_ is a _n x c_ matrix and _B_ a _c x n_ matrix.

That means we can use the BLAS function [GEMM](https://en.wikipedia.org/wiki/Basic_Linear_Algebra_Subprograms#Level_3). For that we should be able to find a suitable [fast implementation](https://en.wikipedia.org/wiki/Basic_Linear_Algebra_Subprograms#Implementations).

Notably:
* Scipy https://docs.scipy.org/doc/scipy/reference/linalg.blas.html#blas-level-3-functions
* Intel MKL https://software.intel.com/en-us/mkl-developer-reference-c-blas-level-3-routines
* CUDA http://docs.nvidia.com/cuda/cublas/index.html#cublas-lt-t-gt-gemm
* ATLAS http://math-atlas.sourceforge.net/
* OpenBLAS http://www.openblas.net/

uellue, 2018-02-07:
In essence, we need a fast [out-of-core](https://en.wikipedia.org/wiki/External_memory_algorithm) GEMM implementation that is optimized to read from HDF5.

uellue, 2018-02-08:
I've been playing with compiling numpy against OpenBLAS and Intel MKL and using the dot product as described above. It doesn't perform well at all (up to 3x slower than naive) when the matrices are very slim (less than 16 masks) in one direction and fat in the other. Multithreading brings a performance penalty under these circumstances and doesn't use the available cores. @sk1p will verify if this happens in a naked C version as well or if it is a numpy thing. Weird!

uellue, 2018-02-14:
Here's the result from OpenBLAS. It performs well with larger mask counts and stack heights. A single mask destroys performance. [dieter-OpenBLAS.xlsx](https://github.com/LiberTEM/LiberTEM/files/1723395/dieter-OpenBLAS.xlsx)

[image]

And this is Intel MKL with similar results [dieter-mkl.xlsx](https://github.com/LiberTEM/LiberTEM/files/1723394/dieter-mkl.xlsx):

[image]

uellue, 2018-02-14:
For comparison the manual version discussed in issue #11 
[result2.xlsx](https://github.com/LiberTEM/LiberTEM/files/1723424/result2.xlsx). Conclusion: OpenBLAS and MKL have potential for improvement for very slim rectangular matrices. The algorithms ´discussed in #11 work better for that case. Remaining question: Is this numpy-specific or does it happen in C as well?

[image]

uellue, 2018-02-14:
Results by @sk1p show the same effect when using OpenBLAS in C. We will use BLAS for now and advertise a master thesis to integrate an improved algorithm in suitable BLAS implementations. That way it will be most useful and flexible.

sk1p, 2018-02-21:
There was one major bug in our benchmarks that prevented them from being comparable: the C BLAS implementation used doubles, where the Python version used `uint32`. Now, with `float64`, the numpy version is almost as fast as the C BLAS version, maybe fast enough for our purposes? For a single mask, the worst case in the numpy version is still 3 times slower than the equivalent C BLAS case though. This may be fixable in numpy, or if we change the way we feed data to numpy, ...

Heatmap for numpy w/ float64:
[image]

Raw results: [fastdot-numpy.xlsx](https://github.com/LiberTEM/LiberTEM/files/1743443/fastdot-numpy.xlsx)

Heatmap for C BLAS:
[image]

Raw results: 
[fastdot-openblas.xlsx](https://github.com/LiberTEM/LiberTEM/files/1743446/fastdot-openblas.xlsx)

Another problem is that our results don't currently include the largest frame size, as our buffer is too small to fit 4 frames, which is our minimum stack height. But that affects both versions so results should be comparable now.

uellue, 2018-02-28:
In order to get a better understanding of the observed inefficiencies in hyperspy/hyperspy#1840, here's an implementation of the benchmark that uses dask for tiling and stacking. Long story short, it is exceptionally fast. The poor performance for low stackheight and maskcount comes from the known poor performance of the used BLAS library. Optimal values are reached when the input chunks fit into the L3 cache together.

[image]

[dieter-dask-OpenBLAS-tiled.xlsx](https://github.com/LiberTEM/LiberTEM/files/1766540/dieter-dask-OpenBLAS-tiled.xlsx)

sk1p, 2018-02-28:
Here is the corresponding dask diagnostics output, for maskcount=16, framesize=512^2, tilesiez=256^2, stackheight=8:

The graph built:
[image]

Profiling output:
[image]

uellue, 2018-02-28:
One could use the graph as a textbook example for an optimized matrix multiplication!

sk1p, 2018-03-05:
As we are gravitating towards using Apache Spark, I rewrote the same benchmark in Scala using [DenseMatrix](http://spark.apache.org/docs/2.2.0/api/java/index.html?org/apache/spark/ml/linalg/DenseMatrix.html) from `spark.ml`. For comparison I included an C OpenBLAS heatmap on the left. The left axis is "framesize-maskcount"

[image]

To summarize: the Scala `DenseMatrix` version performs as well as the C OpenBLAS implementation. This is still all local computation, it will get a little slower when distributing it via Spark.

sk1p, 2018-03-05:
The code for the Scala/Spark benchmark is now [available in the repository.](https://github.com/LiberTEM/LiberTEM/tree/master/benchmarks/fastdot/Scala/sparkfastdot1)

sk1p, 2018-05-22:
The mathematical formulation from the original posting by @uellue should end up in our documentation, when that is done I think this issue can be closed.

PR #97 (2018-08-31): Describe mathematical operation for virtual detectors
Extended the documentation with info from Issue #14. Closes #14.
commit 5b995b89 (2018-08-31): Describe mathematical operation for virtual detectors

Extended the documentation with info from Issue #14. Closes #14.
closed 2018-08-31 by sk1p