LiberTEM/LiberTEM #16: Frameworks to distribute computation on many workers

uellue, 2018-02-23:
Here we can build a collection of frameworks and tools that are used to distribute computation on many nodes. It seems that many, many people had similar ideas to solve similar problems. In the discussion we should focus on frameworks that are in widespread use and are actively maintained.

[Apache Spark](https://spark.apache.org/)
[Dask](http://dask.pydata.org)
[Hadoop](http://hadoop.apache.org/)
[MPI](https://en.wikipedia.org/wiki/Message_Passing_Interface)
[Tensorflow](https://www.tensorflow.org/)

Which ones should we add?

uellue, 2018-02-26:
[Chapel](https://chapel-lang.org/)

uellue, 2018-02-26:
More requirements:
* Easy to deploy on a regular PC
* Reasonable to deploy on existing HPC clusters
* Development know-how available
* Reasonable to build a DIY cluster based on it
* Support sharding of large files: Portions are held in local storage on each node

sk1p, 2018-02-26:
[Apache Flink](https://flink.apache.org/) - a bit like Spark, with a with a focus on stream processing, but also supports batch processing.

sk1p, 2018-02-26:
on Spark / Flink:

* In combination with HDF5:
https://www.hdfgroup.org/downloads/spark-connector 
https://support.hdfgroup.org/pubs/papers/Big_HDF_FAQs.pdf
So there is a beta HDF5+Spark connector, but it is not publicly available (yet?) - using HDF5 with HDFS appears to be quite painful right now.

* Both Spark and Flink support stream processing, so we can update a GUI with new results.
* As I understand, Flink is more suited to re-use intermediate results and iterate on data, where Spark will create new tasks for working on the intermediate results
* Deploying Spark/Flink on a local PC, and integrating that into an easy to install package is a challenge I think. Operating Spark/Flink as a cluster would ideally be something provided by the HPC administrative staff. Setting up a DIY cluster for experimentation should be easy, using the respective [standalone cluster](https://ci.apache.org/projects/flink/flink-docs-release-1.4/ops/deployment/cluster_setup.html) modes.

uellue, 2018-02-26:
# What is the problem, actually
Express a data-intensive algorithm in simple terms and then parallelize and execute it efficiently on a distributed-memory cluster with distributed, chunked storage.
Two extremes:
1. Express the algorithm in a numpy-like fashion and let some middleware handle the details of scheduling, IO, IPC and processing – hyperspy + dask
    * Very simple
    * Easy transition from single-CPU in-core processing to distributed out-of-core processing
    * Can handle problems that are too complex for a human to formulate explicitly
    * The middleware can offer error-handling and recovery
    * Difficult to debug and profile with many abstraction layers stacked on top of each other
    * Distinguishing a very stupid implementation from a very smart one requires a detailed understanding of the underlying execution engine
    * The middleware adds CPU cycles
    * The solution calculated by the middleware might be far inferior to a solution that a human could realistically develop
    * The possibilities might be limited by the operations that the middleware allows
2. Express the algorithm, IO, IPC explicitly parallel with MPI and OpenMP
    * In principle, everything can be implemented
    * Costly operations are immediately visible because they have to be programmed explicitly
    * Easier to debug and profile because of a lower abstraction level
    * Low overhead
    * Parallelizing and distributing the code requires a lot of extra development and might introduce bugs
    * The parallelization and distribution might have to be changed depending on the architecture and characteristics of the underlying system
    * Error-handling and recovery have to be developed explicitly
    * Implementing a good solution for complex computations might in practice be beyond the capabilities of a regular developer
    * A manual solution might be far inferior to the result of a good optimizing middleware

## Results and thoughts so far:

Hyperspy + dask does calculate correct results, but adds a forbidding performance penalty compared to a numpy in-core version even for computations that are trivial to parallelize, chunk and distribute for a human developer. 

A lot of incompatible projects exist that work on solving this issue somewhere between the extremes. 

Using MPI for non-trivial tasks can be painful and inefficient to develop. Changes in an algorithm can amount to rewriting a lot of code.

The Apache Hadoop ecosystem seems to do a reasonably good job if configured correctly. The ecosystem offers a lot of different tools and frameworks that build on a common infrastructure. It seems to have little intersection with numpy and HDF5, which makes it quite incompatible, unfortunately. Installing it on a regular HPC cluster is probably not so easy in practice. The system is designed for deployment on the cloud. 

Chapel tries to allow a seamless transition between the two extremes, defaulting to automatic solutions but allowing to specify explicit directives as integral part of the language. However, it currently has poor IO functions, poor integration with libraries and other languages, a small user base, and less-than-stellar performance according to online resources.

Question: Is there a system/middleware that offers a good balance?

uellue, 2018-02-26:
[Link to Spark machine learning](https://spark.apache.org/docs/2.2.0/ml-guide.html)

uellue, 2018-02-26:
Requirement: Offer a good selection of optimized routines for linear algebra and machine learning, or a middleware that is exceptionally good at optimizing such operations when they are implemented at a higher abstraction level.

uellue, 2018-02-27:
HDF5 in a Hadoop context: https://stackoverflow.com/questions/28565970/reading-hdf5-files-in-apache-spark

It seems difficult because Hadoop needs to connect the index range within the data set to offsets within the file in order to figure out which nodes have which parts of the file in fast access. HDF5 has a complex internal structure so that this mapping requires lots of HDF5-internal information.

uellue, 2018-02-27:
Looking a bit into Lustre, it doesn't seem to have the Hadoop functionality of keeping data directly on the client node to get massive collective IO bandwidth very cheaply. Lustre relies on having its storage on dedicated nodes and transferring through network in a configuration where each client has several fast connections to various storage nodes.

uellue, 2018-02-27:
Essentially, it seems to boil down to two options for good scaling that each have pros and cons.

# HDF5, MPI and Lustre
++ Works on regular HPC clusters
++ Seems to be reasonable to deploy on a laptop with a normal file system instead of Lustre
++ HDF5 file format accepted in community, easy to support
++ Relatively clear path from a single-thread Python/numpy implementation, to a parallel version with MPI, to a C implementation with MPI
-- Needs a lot of expensive hardware to get good collective read bandwidth with Lustre for our application
-- MPI increases development burden

# Apache Spark/Hadoop ecosystem
++ Nice ecosystem with various compatible tools
++ Looks like a comfortable programming model and development platform
++ Hadoop file system gives massive collective read bandwidth with very cheap hardware for our use case
++ Resilience of data storage and computation built-in
++ Easy to deploy on commercial cloud infrastructure
?? Deployment on a laptop unclear. Looks like quite a complex system.
-- Pretty incompatible with the way regular HPC clusters are run
-- Incompatible with numpy and HDF5, and consequently NeXus. Would need files in native Spark/Hadoop file formats and data structures. Conversion with Apache Arrow for numpy/hyperspy.
-- While a Python frontend exists, it is just a wrapper for Java. We don't have much Java development in our community.

uellue, 2018-02-27:
In my feeling, the first choice looks like an easier path with a short-term perspective, but we would start running into limits in the medium and long term when we start working on more complex algorithms. Our code base will become quite messy when we add resilience or complex message passing schemes. Making it run at optimal performance will be a struggle on each cluster, matching HDF5 chunk sizes with Lustre stripes and such fun.

The second choice is quite painful with a short-term perspective because we lose all existing code and HDF5. On the long run we'd probably make it easier to implement parallel machine learning and complex algorithms on very large data sets with reasonable performance balanced with reasonable development cost on clusters or the cloud.

uellue, 2018-02-28:
An example what work with Apache Spark et al would look like: https://de.slideshare.net/KevinMader/interactive-scientific-image-analysis-using-spark

uellue, 2018-02-28:
On a single node, dask is actually doing very well under good conditions: https://github.com/LiberTEM/LiberTEM/issues/14#issuecomment-369186198 The problems in hyperspy/hyperspy#1840 are not a fundamental dask issue.

In general, the first step should be adding RAM and NVMe SSDs to a single node in order to max out performance. That is very cheap, with 20000 € you can get a very decent system. Dask would likely do pretty well on such a system IF it is optimized and used correctly like in the above case.

To scale beyond a single node, a nice cluster like JURECA CAN give you higher collective IO bandwidth than a maxed-out single node, and of course much more storage, but only with very expensive network and storage infrastructure for fast network communication between clients and storage nodes. Dask.distributed could work reasonably well, but not stellar, on such a cluster that allows fast IO on any part of a file on any node.

The advantage of Hadoop over dask.distributed is its strong support for data locality. Storage is kept directly on the computation nodes, and tasks are distributed to those nodes that hold the data. At the same time, the Hadoop file system virtualizes this distributed storage and allows random IO through the backplane if necessary, just with a severe performance penalty.

Hadoop would allow you to go much beyond the JURECA model simply by adding more of the 20k€ nodes and connecting them with a normal network backplane of a rack. With about 200,000 € you could get a system that matches JURECA in throughput while being much cheaper to operate, maintain and program.

In that sense, using the Hadoop ecosystem makes it much easier to scale a solution from a laptop to a workstation to a cluster. Cost scales nearly linear with performance. For clusters like JURECA, costs grow with something like a square law because so much extra high-performance network and storage equipment is needed to get good IO performance between node and storage.

For cloud computing, the cost model looks even better. The price difference between running one node for a week vs. 200 nodes for an hour is not THAT big. One could get processing on demand on very fast systems for an affordable price.

uellue, 2018-03-01:
Apache Spark as part of the Hadoop ecosystem looks like the system of choice for analytics and  machine learning on very large datasets. From what we can see, it has everything we need. @sk1p is working on implementing our benchmark with it to confirm that we get a decent single-node performance for our algorithm. Our plan for HDF5 integration is to develop a consistent way to map HDF5/NeXus data onto [Apache Parquet](https://parquet.apache.org/) and vice-versa so that we can easily convert between the two as needed. We'll see what will be more popular in the future. For very large data sets I'd expect Apache Parquet. With [Apache Arrow](http://wesmckinney.com/blog/high-perf-arrow-to-pandas/) there's a way to load data from Apache Parquet efficiently into numpy.

uellue, 2018-03-01:
Comparison of an Apache Spark solution with our [requirements](https://github.com/uellue/opixtem/wiki/Requirements): https://github.com/LiberTEM/LiberTEM/projects/1

Only requirements that relate to the data processing and storage are considered for GUI and live acquisition.

sk1p, 2018-04-16:
While many observations regarding the requirements for distributed computing still stand, the performance of Spark is not satisfactory: for our use case, most of the time is spent doing copies that are sadly inherit to the JVM approach, even when using raw binary files as the intermediate file format. Adding parquet or Avro on top hurts the performance even more: these formats are built for a different use case, that is efficient filtering and projection (that is, limiting the data read to a subset of columns). The details of the analysis are in issue #18. 

We are now again looking at [dask.distributed](http://distributed.readthedocs.io/en/latest/) as an alternative, using the [`Future` API](http://distributed.readthedocs.io/en/latest/api.html#future) to express our 'work packages'. We still want to exploit the data distribution and locality afforded by HDFS, so the new solution needs to work well on top of HDFS.

sk1p, 2019-01-15:
There is a new interesting project:

https://github.com/mars-project/mars

How this relates to dask:

https://github.com/mars-project/mars/issues/102

Maybe interesting to keep on our radar...

uellue, 2019-01-15:
@sk1p oh nice! I'd like to see how it plays out in comparison to dask. No mention of futures or data locality, though. Let's see! :-)

sk1p, 2019-01-15:
As I understand it, it doesn't work on Python functions as a basis, but on expressions on tensors. That means it can do fancy stuff like fusing expressions together (ála numexpr) where possible, _but_ it also means it can't interop with existing code as much (especially calling other languages). They are using promises as an async primitive, which are, as a concept, very similar to futures, sometimes used interchangeably. But I guess here the computation represented by the promise is an expression, not a python function.

Data locality is not mentioned, true. See https://github.com/mars-project/mars/issues/105 for some discussion about I/O. I guess one could use our planned caching system also in a system like mars, _if_ it allowed to run some kind of adapter on the worker nodes.

uellue, 2021-09-06:
@matbryan52 your Apache Ray prototype would fit into this Issue! :-)

uellue, 2023-10-25:
The current set of executors seem to work pretty well. I think we can close here for now!

closed 2023-10-25 by uellue