LiberTEM/LiberTEM #17: Data representation in Spark

sk1p, 2018-03-06:
We need to find a way to represent our input data and 
* Scientific RDD / SciTensor by [SciSpark](https://scispark.jpl.nasa.gov) ([Github](https://github.com/SciSpark/SciSpark))
  - pro: appears to be a nice interface for working with multidimensional data
  - con: project seems to be dead?
* [Distributed Multi-dimensional Array Toolkit (DMAT)](http://ieeexplore.ieee.org/document/8085038/)
  - con: project doesn't seem to be alive/released yet, or is not open source
  - question: is it limited to 3D volumes of scalars, or can the signal be multi-dimensional, too?
* use Spark's [`ImageSchema`](https://github.com/apache/spark/blob/master/mllib/src/main/scala/org/apache/spark/ml/image/ImageSchema.scala) for each detector frame
  - pro: it's built into spark, so it provides a common interface for other operations; also focus on OpenCV interoperability
  - con: not all pixel formats are supported yet (example: [float32](https://issues.apache.org/jira/browse/SPARK-22730))
  - con: maybe not optimal for performance? one reason our benchmarks run so fast is because they process a large stack of frames at once; looking at the heatmap we start to see very good performance at 32+ frames processed at once

* use more low-level abstractions, for example Spark's `DenseMatrix` (local only, that is, one `DenseMatrix` would only contain a part of the input data local to one node) or `BlockMatrix` (distributed matrix, split into blocks of submatrices)
   - pro: [well-optimized](https://github.com/LiberTEM/LiberTEM/issues/14#issuecomment-370370636) operations that profit from native BLAS implementations

uellue, 2018-03-06:
In general, it seems that [DataFrames](https://spark.apache.org/docs/latest/sql-programming-guide.html) are the preferred way to manage large structured data sets. They correspond more or less to database tables. They allow [complex nested and flexible schema definitions](https://docs.databricks.com/spark/latest/dataframes-datasets/complex-nested-data.html) for each column and can be loaded and saved seamlessly as [Parquet](https://parquet.apache.org/documentation/latest/) file. In principle, one cell of the table can hold an entire detector frame or arbitrarily complex structured data. The matrices/representations for our algorithm(s) would be constructed from such a DataFrame as needed, for example through SQL queries or other filters. With Apache Spark's concept of [RDDs](https://www.tutorialspoint.com/apache_spark/apache_spark_rdd.htm) and [lazy evaluation](https://stackoverflow.com/questions/38027877/spark-transformation-why-its-lazy-and-what-is-the-advantage), these working representations would always be linked to their origin from a Parquet file through their lineage graph and then loaded/constructed/cached as needed.

uellue, 2018-03-22:
[An interesting article about Parquet and Arrow](https://www.kdnuggets.com/2017/02/apache-arrow-parquet-columnar-data.html)

sk1p, 2018-03-22:
Something to keep in mind: [reading from a Parquet file is CPU-Intensive](http://db-blog.web.cern.ch/blog/luca-canali/2017-09-performance-analysis-cpu-intensive-workload-apache-spark) - even with compression disabled, reading parquet is quite expensive. I'll add a test case that demonstrates this to the repository later.

Parquet is really good at only reading the data it needs - which is nice for workloads that filter the data before processing it. Using predicate pushdown, filters from SQL are directly applied to only read the parquet partitions that can possibly contain relevant data. Doing a full scan and reading/decompressing/decoding the whole file is the worst case.

A possible direction would be to use simpler files for storing the detector frames. For example, storing stacks of raw frames as contiguous files of doubles, or using Apache Avro (according to [some](http://kitesdk.org/docs/1.1.0/Parquet-vs-Avro-Format.html) [sources](https://stackoverflow.com/questions/28957291/avro-v-s-parquet) Avro is more suited for the full scan case).

There is also [the flare project](https://flaredata.github.io/2017/04/14/parquet-on-fire.html), that includes a sped-up version of parquet, but it seems to be proprietary at least for now. They do some really fancy partial evaluation / JIT to gain some impressive speed-ups.

uellue, 2018-03-22:
Hmm, having efficient selection is actually nice to construct blocks with a local environment based on a query on data with complex scan pattern.

sk1p, 2018-04-16:
Closing as we are no longer considering apache spark as a possible solution

closed 2018-04-16 by sk1p