LiberTEM/LiberTEM #19: Intermediate format for distributed processing

sk1p, 2018-04-16:
While we still want to use NeXus as our exchange file format (and as input format for single-node processing), we think that HDF5-based formats are not suitable for use together with HDFS. This is because HDF5 doesn't have native support for HDFS, and using them together creates some ugliness and inefficiencies:

 * Can't open HDF5 files from a file descriptor. This is important for short circuit reads on HDFS
 * HDF5 is not aware of HDFS blocking and vice versa. This means splitting the HDF5 file into smaller HDF5 files 'by hand' is needed and some kind of layer for translating global coordinates to the local partitions.

So we need a file format on top of HDFS that fulfills the following requirements:

1) Efficient use together with [HDFS](https://hadoop.apache.org/docs/r1.2.1/hdfs_user_guide.html) for data locality - this is our main objective!
   * This includes opening files from file descriptors for [HDFS short-circuit reads](https://hadoop.apache.org/docs/current/hadoop-project-dist/hadoop-hdfs/ShortCircuitLocalReads.html)
2) Efficient bulk reading without any extra copying or byte order swapping
   * in the case of compressible data it may be preferable to allow a decompression step if a) not too much CPU is used, and b) the compression output rate is significantly higher than the usual raw I/O read speed
3) Handle different pixel data types (`float32`, `float64`, `uint`, ...)
4) Comfortably handle multi-dimensional data - work with `numpy` and provide a nice slicing abstraction

The first three are absolute *must* IMO, the last one is nice to have, but not crucial at the I/O layer (as long as parts of the file can be read as numpy arrays)

sk1p, 2018-04-16:
Edited to add 5)

sk1p, 2018-04-16:
Our current thinking: we should support both TileDB and Raw Binary Files as intermediate format, and HDF5/NeXus as local input format. When TileDB gets as fast as using raw files (or maybe 80% there), which hopefully coincides with us moving to more complex access patterns, we can drop support for raw files. Thoughts?

uellue, 2018-10-02:
A quick update about the experience so far.

1. Memory mapping in combination with the FS cache is the fastest way to get a file into the CPU. Blocking etc. should ideally be left entirely to the data processing function, for example torch. This already gives local, compact access patterns if the processing is properly optimized. The OS does an excellent job with paging in such cases. If a file is already in the FS cache, mmap seems to essentially give a pointer directly to this cache, which avoids any copy operation -- at least on Linux.
2. A future file format should therefore be essentially a set of function calls that are compiled and _inlined_ in the processing routine to access the contents of the mmapped file. An mmapped raw file fed directly to a numerics library is a trivial example for that: Logical addresses and numbers in the "computation space" translate very directly to memory addresses and memory content, which in turn correspond to file offsets and file contents. A more complex file format would include a transformation step for addresses and contents. If that is compiled into the processing kernel, the optimizer can co-optimize address translation and decoding with the actual calculation. That avoids any intermediate buffers and memory transfers.

sk1p, 2019-12-02:
A possible container format to support compression for our caching layer (see also: #136): Caterva

Repo: https://github.com/Blosc/Caterva
Docs: https://caterva.readthedocs.io/en/latest/
Intro Slides: https://blosc.org/docs/Caterva-HDF5-Workshop.pdf

Python binding: https://github.com/Blosc/cat4py
Docs: https://cat4py.readthedocs.io/en/latest/

"In addition, Caterva adds machinery for retrieving arbitrary multi-dimensional slices (aka hyper-slices) out of the multi-dimensional containers in the most efficient way. Hence, Caterva brings the convenience of multi-dimensional and compressed containers to your application very easily."

We could create a separate Caterva array for each partition. This would "trivially" give us support for tiling for large frames (#27, #373, related: #331, ...), while also tackling our need for compression (see for example #387).

uellue, 2019-12-02:
Nice! This one looks promising. :+1:

uellue, 2023-10-25:
Raw binary, NPY and possibly Zarr seem to work pretty well for now. Closing!

closed 2023-10-25 by uellue