LiberTEM/LiberTEM #20: TileDB

sk1p, 2018-04-16:
The TileDB project looks like a good fit as an intermediate "file format" (see #19). There are some inefficiencies currently (see [TileDB-py issue 38 for details](https://github.com/TileDB-Inc/TileDB-Py/issues/38), also [527](https://github.com/TileDB-Inc/TileDB/issues/527) and [526](https://github.com/TileDB-Inc/TileDB/issues/526) on the parent TileDB project), but they are not inherent to the platform (like the copying with JVM-based solutions).

 * [Basic concepts](https://docs.tiledb.io/en/stable/tiledb101/basic-concepts.html)
 * [Python API](https://docs.tiledb.io/projects/tiledb-py/en/latest/python-api.html)
 * [Storage Backends and HDFS](https://docs.tiledb.io/en/latest/tiledb101/storage-backends.html#hdfs)

TileDB allows to alter the order in which the data is written to disk, to allow efficient processing for different access patterns (linear (row/col-major), local environments, ...). This fits very well with #11 where we figured out the benefits of applying the right tiling to the input data at the computation level.

sk1p, 2018-04-16:
[The TileDB paper](https://people.csail.mit.edu/stavrosp/papers/vldb2017/VLDB17_TileDB.pdf)

sk1p, 2018-07-12:
As an update, TileDB 1.3.0 was [recently released](https://tiledb.io/v130-released). Data locality with HDFS does not seem to be implemented yet (memmapped I/O is milestoned for the next major release though).

uellue, 2021-02-16:
Closing here since we focus on supporting many existing file formats well. Given the performance of most file formats in LiberTEM and the tiling update in version 0.6, one can consider the underlying issue solved. 🚀

closed 2021-02-16 by uellue