LiberTEM/LiberTEM #26: Optimize binary masks

sk1p, 2018-05-22:
For virtual detector applications, we can work with simple bitmasks.

Right now, binary masks are multiplied with ``np.ones``, with dtype=float/double to create float/double masks for ``dgemm``/``sgemm``. There is potential for optimization, as the masks are now 32 (or 64) times larger than they could be: they could be densely packed into bitmasks!

A possible naive implementation:

```C
#include <stdio.h>

void apply_mask(unsigned char* mask, int* image, int* result, int imageSize) {
    int i = 0;

    for(int m = 0; m < imageSize / 8; m++) {
        char maskbyte = mask[m];
        for(int k = 0; k < 8; k++) {
            int pixel = image[i];
            result[i] = (maskbyte & 0x1) ? image[i] : 0;
            maskbyte = maskbyte >> 1; 
            i++;
        }
    }
}

int main() {
    unsigned char mask[] = {0x0, 0x0, 0x2, 0xFF};
    int image[] = {
        1, 2, 3, 4, 5, 6, 7, 8,
        1, 2, 3, 4, 5, 6, 7, 8,
        1, 2, 3, 4, 5, 6, 7, 8,
        1, 2, 3, 4, 5, 6, 7, 8,
    };
    int result[32];

    apply_mask(mask, image, result, 32);

    for(int i = 0; i < 32; i++) {
        printf("%d\n", result[i]);
    }
    return 0;
}
```
A good compiler should be able to compile this down to a bunch of vectorized ``cmov`` instructions (those are part of SSE or one of the updates I think?) and shifts. The innermost loop can be optimized by hand-unrolling and then using 8 constant masks to extract the bits, instead of shifting the ``maskbyte``. This removes the data-dependency of each iteration of the innermost loop on the previous iteration.

As we are then outside of BLAS territory, we would need to cache-optimize (L2) this ourselves, but this approach has the advantage that even for larger frames, the binary masks can completely stay in L3 cache.

Example: an 2048×2048 bitmask is 4194304 bits, that is 512 KBytes, so on a beefy system with 8MB L3 cache, filling half of the L3 cache will give you 8 masks.

uellue, 2018-05-23:
This sounds like a fun project, especially looking what different compilers do with the C code!

sk1p, 2018-06-07:
It may be of interest to apply the lessons from https://github.com/flame/how-to-optimize-gemm

uellue, 2020-01-22:
This would be related to #555 and could be a scenario where a custom Numba-based GEMM would make sense.

uellue, 2023-10-25:
Closing for now since `ApplyMaskUDF` has never been a limiting factors in scenarios with binary masks. To be reopened when this comes up in the real world!

closed 2023-10-25 by uellue