LiberTEM/LiberTEM #27: Properly handle large frames: sub-frame tiling

sk1p, 2018-05-22:
Right now, performance is suboptimal once a stack of frames is larger than (rough guess, depends on number of masks) half of the L3 cache. To properly handle large frames, we need to create ``DataTile``s that a) subdivide the frames, and b) still stack a few of these sub-frames on top of each other. Example: let's say we have a dataset with shape ``(128, 128, 4096, 4096)``, dtype=float. Each frame then takes up 64MB. On a system with 8MB L3 cache, only a fraction of a frame can fit into memory. Something like ``(1, 8, 256, 256)`` could still fit the L3 cache though.

In #11 we already had a look at the optimal execution order. We need to implement one of the efficient tilings presented there, throughout the system. 

- [ ] sub-frame tiling for our file format(s), basically #20 or a custom implementation for our binary file format
- [ ] properly handle sub-frame tiling in the mask application task
- [ ] proper result handling for sub-frame tiling

On disk the data needs to be in an efficient tiling, otherwise we need to do multiple RAM<>L3 roundtrips for each frame. This part may need to wait until we are using TileDB as our backend, so we are mostly waiting on https://github.com/TileDB-Inc/TileDB-Py/issues/38 there. A remaining question would be: do we want to support this on non-cluster systems? That would mean we need to pre-process even on single node systems (for example have a HDF5 -> HDF5 conversion, or HDF5 -> TileDB, that creates 2MB chunks of shape ``(1, 8, 256, 256)``). Then we would have non-straight-forward chunking for those files...

Result handling should be easy, we basically need to add to the (maybe) existing partial result, instead of only copying into the result array. The ``ResultTile`` needs to be clearly marked as being a partial result.

Comfortably handling sub-frame tiles at the lower level is not so easy, once we go beyond applications that use mask application as their basis. Then we may need to think about boundary handling etc. For now, we need to slice the masks array such that it matches the sub-frame tiling.

sk1p, 2018-06-07:
Sub-frame tiling is now implemented for HDF5 data sources. There is still some room for optimization, for example the buffers for border conditions are not reused (if the tilesize doesn't evenly divide the frame size)

uellue, 2020-01-22:
Refs #554

PR #706 (2020-04-27): WIP: Tiled reading in more file formats
Fixes #554, fixes #634, fixes  #27, fixes #331, fixes #373, fixes #635, fixes #435, fixes #247 

## Functionality implementation

- [x] Update `MemoryDataSet` #634 
- [x] Create API for tileshape negotiation #554 
- [x] Add tiling scheme and current index to `UDFMeta` #635 
- [x] Update all `DataSet` impls to work with the new infrastructure
  - [x] hdf5 - can't use the I/O backend as I/O is done by HDF5 itself
  - [x] ser - I/O is done by ncempy, so we can't use the I/O backend
  - [x] raw
  - [x] mib
  - [x] blo
  - [x] k2is
  - [x] frms6
  - [x] empad [... cut]
closed 2020-04-27 by sk1p