LiberTEM/LiberTEM #28: Cropping/Culling

sk1p, 2018-05-22:
There are two cases of cropping we should support:

1) Cropping to the region of interest of the user, in the scanning dimensions
2) Automatic culling in the frame dimensions

For 1) I think it makes sense to wait until we have user demand and/or a user interface.

For 2) it makes sense to crop on two layers: computation and I/O. Once we have sub-frame tiling from #27 implemented, each tile can hold its bounding box and the I/O layer can check if we are interested in the data from this tile, before even reading the data. Example: bright field only needs data from the center of the frame, so we only read tiles whose bbox intersects with the bright field disk. In the computation layer, we can reduce the number of FLOPS we need to execute by tightly cropping the data (and mask) to the bbox of the bright field disk before executing the matrix multiplication.

sk1p, 2018-08-28:
Cropping the scanning dimension is now supported, even though the implementations for each DataSet subclass are not as efficient yet as they could be.

sk1p, 2018-08-28:
I should say: there is no user interface yet, only low level support. It is currently used by the frame picking implementation.

uellue, 2023-10-25:
Closing since this has become pretty stale, apparently not a big issue. It might come up again with the "lazy slice" idea on file readers.

closed 2023-10-25 by uellue