LiberTEM/LiberTEM #29: Automated comparison against reference implementations

sk1p, 2018-05-23:
For some methods we want to implement, there are existing proven implementations (example: ``scipy.ndimage.measurements.center_of_mass``). It would be useful to have automated tests for these that compare our implementations against reference implementations and that are run as part of CI.

Depends on #25

magnunor, 2018-06-09:
I've been thinking about something like this, however doesn't the CI containers usually have a very low amount of memory and CPU-cores? So the benchmarking would be quite far from what people normally would use on their own computers.

Still, this would definitely be useful, since currently all of this has to be done manually, which is not really a good way of handling these things.

sk1p, 2018-06-11:
In this case, I was thinking more correctness than performance, but having an automated benchmark suite would be nice, too. I agree that it would not run well on public CI infrastructure, but maybe we could in the long run have a private Jenkins instance running somewhere where we would have control over available resources etc.

sk1p, 2018-08-07:
An example how other projects handle this: http://www.astropy.org/astropy-benchmarks/

uellue, 2018-11-22:
Really important because this could have caught #172

sk1p, 2018-11-22:
We now test against the high-level API, and have a comparison in place against the scipy results for COM. This should make things more robust in the future.

closed 2018-11-22 by sk1p