LiberTEM/LiberTEM #31: Visualization and UI for center of mass shift

uellue, 2018-07-09:
* Select rotation and ROI in detector frame
* Separate X and Y component image
* Combined image with suitable vector field visualization

sk1p, 2018-08-06:
See https://github.com/LiberTEM/LiberTEM/pull/52 for the most important part of vector field visualization

sk1p, 2018-08-08:
The vector field visualization is done, rotation and UI adaptions are still TODO

uellue, 2018-10-11:
Deferred rotation to milestone 0.2 since it is usable without that.

sk1p, 2020-03-17:
Our CoM visualization can directly use the empyre package, now that it is published to pypi:
[image]

/cc @jan-car 

Note that the the y-axis of the empyre plot functions is flipped by default, compared to the default mpl `imshow`, need to make sure to keep this consistent!

jan-car, 2020-03-17:
Glad this works! I just added some more functions which I will open a merge request for, in the next minutes.

Concerning the flipped y-axis, that caused much headache for me aswell, especially for some rotation methods... If I understand it correctly, the convention to have the y axis pointing down stems from the times of old television cathode ray tubes which scanned from left to right, top to bottom and stuck around for image processing after that.

Coming from physics I wanted to have my axes pointing in the directions I knew, so x to the left, y upwards. This decision was made at the very beginning of my thesis, so it permeates through most of my functions concerned with plotting.

The `emp.vis.imshow` command relays all additional `**kwargs` to `matplotlib.pyplot.imshow`, can you try to add `origin='upper'` to see if this works? Or do the colors not change accordingly?

If this is too inconsistent, we can try to make `imshow` of empyre and matplotlib consistent, might be some work though.

sk1p, 2020-03-17:
> Coming from physics I wanted to have my axes pointing in the directions I knew, so x to the left, y upwards. This decision was made at the very beginning of my thesis, so it permeates through most of my functions concerned with plotting.

:+1: - maybe this is something we should fix in LiberTEM actually. As we don't yet document the GUI axes, we can take the opportunity to change and document it, maybe also adding indicators in the GUI. Thoughts @uellue ?

> The emp.vis.imshow command relays all additional **kwargs to matplotlib.pyplot.imshow, can you try to add origin='upper' to see if this works? Or do the colors not change accordingly?

The colors stay exactly the same with `origin='upper'`, but the colorwheel can be made consistent by also passing the origin. Only the quiver plot doesn't accept an origin:

[image]

jan-car, 2020-03-17:
Yeah, quiver does not play nice at all, I had a lot of other problems during its implementation. It shouldn't be too hard to implement an `origin` argument, though, maybe that would be a nice test for the  feature request issue you enabled over on the EMPyRe page 😉.

If not provided via `kwargs`, all plotting commands take their defaults from a stylesheet `empyre-image`, which you can either install into your user folder via `emp.vis.copy_mpl_stylesheets` and change there (e.g. `origin='upper'`) or, if EMPyRe does not find them in your user folder, it uses the ones inside the package itself (which will be overwritten when you reinstall it, though).

All that, of course, if decide to stick with the convention of having the origin on the top. I always found it strange, but maybe it is not just a historic artefact. Any input is appreciated.

AyushGit123, 2020-03-23:
Hello Mentor(@uellue @sk1p ),

My name is Ayush Shukla, I am a Second year undergraduate student of Electrical Engineering student at National Institute of Technology Surat, Gujarat, India. I am a Red Hat Certified System Administrator on Red Hat Linux 8. I am quite experienced on Python and have a good hands on using Python libraries.

I want to work on this project during GSOC 2020. Please guide me about the things that I would have to do and what would be the workflow

sk1p, 2020-03-24:
Hey Ayush,

> I want to work on this project during GSOC 2020. Please guide me about the things that I would have to do and what would be the workflow

I replied to your message on gitter, let me know if you need any more support!

AyushGit123, 2020-03-24:
> Hey Ayush,
> 
> > I want to work on this project during GSOC 2020. Please guide me about the things that I would have to do and what would be the workflow
> 
> I replied to your message on gitter, let me know if you need any more support!

@sk1p I had already installed LiberTEM on my PC and since I want to work on this project please tell me which programming languages should I master to be capable of doing this project and where should I start from.
Now I have experience in Python, Java, C language.

sk1p, 2020-03-25:
Copy of the message from gitter:

> @AyushGit123 Hi and welcome to LiberTEM! As a first starting point, I recommend thoroughly reading https://python-gsoc.org/#gettingstarted and https://google.github.io/gsocguides/student/ - they contain a lot of information on how GSoC works. Then, a good way to get started is to pick an issue marked "good first issue" and try to fix it - this will give you a first impression of the LiberTEM code base and our development process. As the deadline for proposal submissions is coming closer, I would advice to start to write the proposal ASAP - and ask any questions you have about the project!

> Also note that our GUI is written in TypeScript/react etc., so if you don't have any experience with those yet, be sure to plan accordingly!

That should answer most of your questions. 

> which part should I be focusing on

This project idea concerns mostly our GUI, so I would suggest to try to fix a simple issue in the GUI to get started, but also I would recommend reading a bit about the physics behind the CoM method to understand the motivation.

uellue, 2021-08-19:
Bumped in priority, CC @bdalevin :-)

PR #1087 (2021-08-24): CoM rotation and flip in the GUI
Fixes #31 

## TODO

- [x] The template for notebook download should also be updated to include the new parameters.
- [x] Fix results data download.
- [x] Add better input method (slider) in addition to allowing to enter values directly
  - [x] Fix for easily entering negative rotation angles in number input
- [x] Add documentation
  - [x]  CoM section in applications section → @uellue will take care of this one
  - [x] Explain parameters (info box or similar?)
- [x] Fix bug: time delta resets after analysis finishes

## Contributor Checklist:

* [x] I have add [... cut]
closed 2021-08-24 by sk1p