LiberTEM/LiberTEM #35: Signal handler issues on Windows

uellue, 2018-07-09:
Win 10 with Anaconda, LiberTEM HEAD

(LiberTEM) C:\Users\weber\Documents\LiberTEM>libertem-server                                                            [2018-07-09 11:13:47,286] DEBUG [asyncio.__init__:65] Using selector: SelectSelector                                    Traceback (most recent call last):                                                                                        File "C:\Users\weber\AppData\Local\Continuum\anaconda3\envs\LiberTEM\Scripts\libertem-server-script.py", line 11, in <module>                                                                                                                     load_entry_point('libertem', 'console_scripts', 'libertem-server')()                                                  File "C:\Users\weber\AppData\Local\Continuum\anaconda3\envs\LiberTEM\lib\site-packages\click\core.py", line 722, in __call__                                                                                                                      return self.main(*args, **kwargs)                                                                                     File "C:\Users\weber\AppData\Local\Continuum\anaconda3\envs\LiberTEM\lib\site-packages\click\core.py", line 697, in main                                                                                                                          rv = self.invoke(ctx)                                                                                                 File "C:\Users\weber\AppData\Local\Continuum\anaconda3\envs\LiberTEM\lib\site-packages\click\core.py", line 895, in invoke                                                                                                                        return ctx.invoke(self.callback, **ctx.params)                                                                        File "C:\Users\weber\AppData\Local\Continuum\anaconda3\envs\LiberTEM\lib\site-packages\click\core.py", line 535, in invoke                                                                                                                        return callback(*args, **kwargs)                                                                                      File "c:\users\weber\documents\libertem\src\libertem\web\cli.py", line 8, in main                                         run(port)                                                                                                             File "c:\users\weber\documents\libertem\src\libertem\web\server.py", line 642, in run                                     loop.add_signal_handler(signal.SIGINT, do_stop)                                                                       File "C:\Users\weber\AppData\Local\Continuum\anaconda3\envs\LiberTEM\lib\asyncio\events.py", line 499, in add_signal_handler                                                                                                                      raise NotImplementedError                                                                                           NotImplementedError

sk1p, 2018-07-09:
Could you re-test, please? Now using the signal facilities of `tornado` instead of `asyncio`

uellue, 2018-07-11:
Yes, it is fixed now! Thx :-)

closed 2018-07-11 by uellue