LiberTEM/LiberTEM #36: Basic File Browser

sk1p, 2018-07-09:
For opening local HDF5 / RAW files

sk1p, 2018-07-09:
Could additionally accept Drag&Drop events - https://developer.mozilla.org/en-US/docs/Web/API/HTML_Drag_and_Drop_API/File_drag_and_drop

uellue, 2018-07-09:
Drag and drop sounds awesome! We should make sure that the entire backend is local and display a helpful message if it is not. I'm quite sure someone will try this with a remote cluster...

sk1p, 2018-07-09:
Yeah, otherwise it could offer to upload & ingest the dropped file, though I don't know how efficient it is to do the upload via the browser. Could actually work...

uellue, 2018-07-09:
Good idea to offer upload! With a confirmation step because it might not be what the user wants to do, for example if the same network file system or cloud folder is mounted on both client and server.

sk1p, 2018-07-09:
Yeah, also the binary formats may need some parameters to be read...

So the flow would be:

1) Drop file
2) Get a parameters dialog
3) If the file is remote, start uploading (with an indicator somewhere on the screen - normal work can continue)
4) If the file is not remote, or on completion of upload, open file as Dataset

The "normal" (non-DnD) flow should be the same actually - you first select a file (from local file system or from HDFS), then you get the parameter dialog, and then it is opened. This enables us to make some things more user friendly, for example we could pre-fill some form fields etc. For EMPAD we can at least guess the scan dimensions from the filesize, and provide information for form input validation, which is a bit awkward if the file name is also part of the same form.

I'll create a separate ticket for uploading via DnD - I think it fits for the 0.2 milestone, do you agree?

sk1p, 2018-08-01:
Actually DnD for direct reading (without 'uploading') will be limited, as the [fullPath](https://developer.mozilla.org/en-US/docs/Web/API/FileSystemEntry/fullPath) property is not universally supported.

sk1p, 2018-08-10:
For the client, we should use [react-window](https://react-window.now.sh/#/examples/list/fixed-size) to render the list of files, to support directories with many files.

Need to think about how to properly open MIB files - idea: instead of specifying a glob pattern, just select the *.hdr file, use its 'basename' to construct a glob: `default.hdr` -> `default*.mib`

sk1p, 2018-08-10:
There are some security considerations concerning the file browser (or even without the file browser: loading from arbitraty paths) and multi-user environments - LiberTEM always has access to all files of the user it is running as - including sensitive information like passwords or SSH keys, and other users can access that LiberTEM instance.

We should think about this a bit more, and at least introduce a cmdline switch to whitelist access to certain paths, where the user expects to load data from, or generate a token which the user has to include when he first accesses the UI.

uellue, 2018-08-10:
Hm, how does, for example, Nextcloud handle authentication? Is there a "best practice" for a system like LiberTEM that gives a web-based interface for local files? Can we assume that LiberTEM is always running as the Unix user that belongs to the "web user" or might there be web users that don't have a local Unix account?I'm just a bit worried because in particular on a cluster it gets messy IMO if we can't rely on the operating system to manage access. We should probably not roll our own solution but rely on good third party code. Is there a code base we could use?

davidelbert, 2018-08-13:
I'm new to these LiberTEM discussions, but want to suggest checking out Globus authentication.  It provides access to SSO for academic and lab users as well as OpenID (Google authentication), and InCommon.  It would also make it easy to tap into a growing materials data infrastructure where Globus is the de facto standard for large file transfer and offers storage solution.  There are lots of resources to learn more, but one place to start would be the Globus docs: [https://docs.globus.org/authorization-authentication-guide/](https://docs.globus.org/authorization-authentication-guide/)

sk1p, 2018-08-14:
@davidelbert thank you for joining the discussion! Globus looks interesting, from a quick glance it looks similar to the [B2ACCESS/B2SHARE/B2DROP services provided by EUDAT,](https://www.eudat.eu/services/) right? I'd love to learn more about that, do you have first-hand experience using this system?

sk1p, 2018-08-23:
"disappearing files" bug filed upstream: https://github.com/bvaughn/react-window/issues/45

uellue, 2018-08-27:
4ca7310 fixed the scrolling issue for me. Thx @sk1p !

commit 4e0435b7 (2018-08-27): client: file browser finishing touches, closes #36
closed 2018-08-27 by sk1p