LiberTEM/LiberTEM #39: Package/Installer for Windows

sk1p, 2018-07-09:
Can the average windows user install a Python environment and install LiberTEM via pip, or do we need to provide an `.exe` installer or something like that?

uellue, 2018-07-09:
Having an EXE would be required sooner or later IMHO since the LiberTEM GUI is becoming useful as a quick and simple browser for pixelated STEM files. Installing and maintaining the environment is probably quite a barrier and simply overkill for most users who just want to have a quick look at some files.

sk1p, 2018-07-12:
I see at least two ways for this: 

1) build a [portable miniconda installation](https://github.com/conda/conda/issues/1977) and create an installer that just copies that somewhere
2) like hyperspy, create [a bundle based on WinPython](https://github.com/hyperspy/hyperspy-bundle)

I currently like 1) as it is basically what I'm doing with the Linux app image

uellue, 2018-08-22:
Actually, installing Anaconda and then installing LiberTEM into an environment is quite OK on Windows. Maybe we can stick with that for the early version since people likely want to update from Git anyway?

AshwinB-hat, 2019-03-29:
Hey @uellue @sk1p,
I would like to work on the conda packaging for windows.

uellue, 2019-03-29:
@AshwinB-hat that's great! I suppose as part of the Google Summer of Code?

We can discuss details in the Gitter chat: https://gitter.im/LiberTEM/Lobby

AshwinB-hat, 2019-03-29:
Yes! 
Fine i will ping there. (Y)

uellue, 2024-04-12:
...or this https://pypi.org/project/pynsist/

uellue, 2024-04-17:
Pynsist managed to build an installer package that installed sth. However, when trying to run it, it would flash a window and terminate. I couldn't get a proper error message, so I can't continue investigating.

PyInstaller is currently broken by this: https://github.com/open-telemetry/opentelemetry-python/issues/2288

uellue, 2024-04-17:
https://github.com/pyinstaller/pyinstaller/issues/8413