LiberTEM/LiberTEM #40: Package/Installer for Mac OS X

sk1p, 2018-07-09:
Same question as #39 but regarding Mac OS X - we will also need testers for this.

uellue, 2018-08-22:
Same as Windows #39: Installing from source into a Python environment should work pretty well for now. I didn't try it personally, though.

sk1p, 2018-08-23:
I guess it should be fine - for later, my plan was to build an App with py2app, but I don't know how much work that actually is. Also, I don't know if there is a "run in terminal" mode that can be somehow enabled for an App.

sk1p, 2018-08-27:
For reference: https://github.com/Pext/Pext/blob/master/travis/build-dmg.sh

sk1p, 2020-02-10:
Removed the milestone for now - should be implemented if we have concrete user demand (and someone to do QA on Mac)