LiberTEM/LiberTEM #46: Export calculation result

uellue, 2018-07-16:
Currently, the GUI shows a false color map of the calculation result.

It should be possible to export the result as numerical values, for example as raw binary, HDF5 or CSV so that it can be used for further processing.

sk1p, 2018-07-16:
Digital micrograph supports a weird CSV format, which is just rows of intensity values separated by commas - we should also export to this dialect, in addition to the more straight forward "x, y, intensity1, ..., Intensity_n"

Maybe also float .tiff images?

bdalevin, 2020-02-26:
I would like to add a comment to this issue. 
Allowing users to save a float .tif image from the GUI would be extremely helpful for many of my users. 
Please implement this as soon as you can.

sk1p, 2020-02-27:
Thank you for the feedback! We will prioritize this issue and plan to implement this for the 0.5 release.

closed 2020-03-18 by sk1p

sk1p, 2020-03-18:
The basic functionality is now implemented, with support for HDF5, NPZ, TIFF, and RAW formats. It can be tested by [installing LiberTEM from git](https://libertem.github.io/LiberTEM/install.html).

If you have any feedback, let us know in this issue, the [followup issue](https://github.com/LiberTEM/LiberTEM/issues/687) or via E-Mail. Thanks!