LiberTEM/LiberTEM #59: Feature request: Automated STEM-diffraction analyses for orientation and phase mapping

woozey, 2018-08-08:
Establish analyses similar to what ASTAR does:
http://www.nanomegas.com/Articulos/Nanomegas/Products_Tem_orientation_imaging.html
i.e. simulate diffraction patterns for different orientations and materials (material data to be provided by user), find best matching patterns for each scan point.

sk1p, 2018-08-08:
I wonder how that can be done efficiently, i.e. without "brute-force" comparison to each pattern in the database of simulated patterns. I'm thinking about https://en.wikipedia.org/wiki/Geometric_hashing or something along the lines, if this can be applied here. For an interesting use of this technique, see this talk about detecting stars in the night sky: 

http://cosmo.nyu.edu/hogg/research/2006/09/28/astrometry_google.pdf

My question would be: what are the (geometric) parameters that we can detect and use as distinguishing factors?

woozey, 2018-08-08:
For the simulation part, I also will have to look into it, though I can imagine that it can be relatively straight forward depending on level of detail that one will aim to reproduce in simulations...

uellue, 2018-08-08:
> Typically diffraction patterns are indexed (identified) by measuring distances and angles between spots. This certainly can be done using geometric hashing or some other computer vision method (I'm not very familiar). We may need to find people who has better crystallography background than me in order to find appropriate way of interpreting this information in terms of orientation and etc. I'll try to make a small research on which codes are already available and find people who might be interested to help.

The number of possible symmetries in a crystal, and consequently the different families of possible diffraction patterns, is rather limited and pretty much completely solved with some advanced mathematics. https://en.m.wikipedia.org/wiki/Space_groupEach family of possible crystal lattices has a well-defined list of free parameters, at most three independent 3D vectors, that define where spots can potentially be found. Many common materials have much less degrees of freedom, namely cubic lattices. What intensity these spots have, how large they are and any background intensity are then a different topic.For a cubic material, lattice constant, information about incident wavelength, camera length, detector pixel size, orientation of the crystal in the beam (zone axis and rotation) are completely sufficient to say exactly where spots can be found. Some of them might have zero intensity, though.In that sense, there's a whole theory framework and (afaik) lots of existing open source code about classification and indexing of diffraction patterns.

magnunor, 2018-08-08:
pyXem has a lot of functionality for analysing precession data: https://github.com/pyxem/pyxem