LiberTEM/LiberTEM #61: CPU load from libertem-server when idle

uellue, 2018-08-10:
Observed behavior:
The libertem-server process creates significant CPU load on the moellenstedt machine even when no processing is performed and the GUI window is closed. The load seems to increase with the number of workers -- 4 cores on @sk1p 's machine not noticeable, 18 cores on moellenstedt significant.

Expected behavior:
LiberTEM only creates CPU load when it is doing useful work.

Suspicion:
The dask scheduler might be the culprit (@sk1p )

sk1p, 2018-08-10:
See also: https://github.com/dask/distributed/issues/2079

uellue, 2018-12-13:
I tested with distributed==1.25.0 released about two weeks ago, and libertem-server still creates significant background load (one process with about 10 % CPU usage) on moellenstedt as soon as the cluster is running. It produces the same load with distributed==1.24.0 which was released before merging dask/distributed#2339.

In conclusion, I'd say that this change didn't improve things for us. 10 % still feels like a bit much for an idling cluster.

sk1p, 2019-06-25:
There is actually a new workaround posted on the dd issue: https://github.com/dask/distributed/issues/2156#issuecomment-503735503 - we just have to try it and then figure out the right way to set the configuration. It can be done with [dask.config.set](https://docs.dask.org/en/latest/configuration.html#directly-within-python), but it kind of feels dirty to just globally set the dask config when creating an executor (or when starting a local cluster?)

sk1p, 2020-12-14:
I just re-checked the idle CPU usage. Just running `libertem-server` and creating a local cluster creates 15-20% idle usage on the main process, and additionally 2-3% per worker process.

Setting all four environment variables as in #921 reduces idle load to about 12% in the main process, and on average no load at all per worker process. 

```sh
DASK_DISTRIBUTED__ADMIN__TICK__INTERVAL=1000
DASK_DISTRIBUTED__SCHEDULER__WORK_STEALING_INTERVAL=1000
DASK_DISTRIBUTED__WORKER__PROFILE__CYCLE=60000
DASK_DISTRIBUTED__WORKER__PROFILE__INTERVAL=30000
```

uellue, 2023-04-20:
Checking on Ptycho, the idle load with default settings is at 10 % for the main process and about 1 % for the workers. The environment variables above lower it to 6 % for main and 0.3-0.7 % for workers.

On a quad-core Windows machine the load is very low as well.

Closing since this doesn't seem to be a big issue anymore.

closed 2023-04-20 by uellue