LiberTEM/LiberTEM #64: Stopping server takes too long

sk1p, 2018-08-15:
Stopping the server currently takes up to 10 seconds, which is too long. There are some timeouts but I'm not sure what dask/asyncio/tornado/... is waiting for

sk1p, 2018-09-12:
This should be a little bit better now, I simply reduced the timeout, as it was strangely always hit (even when set to the default 10 seconds?). I did not see any negative consequences yet (lingering processes or something like that), but we should keep an eye on that.

@uellue could you test this on Windows, if you get the chance? Stopping should now take a little longer than 1s.

uellue, 2018-09-12:
@sk1p can do, when I'm back in Jülich. I ran into issues here with the Win7 installation.

uellue, 2018-09-26:
It works now. The server stops and calculation ends within 2-3 seconds. :-) 👍

closed 2018-09-26 by uellue