LiberTEM/LiberTEM #65: Automatically select parameters based on selected file

sk1p, 2018-08-15:
Currently the user needs to set a few parameters, depending on the file format. We can do better and do auto detection. The workflow then looks like this:

 * User selects a file in the file browser
 * LiberTEM server tries to detect the format and parameters
 * LiberTEM client shows open form pre-filled with detected parameters
 * User can still change parameters and even select different format
 * User confirms selection and LiberTEM opens the dataset

sk1p, 2018-10-11:
This should be considered high-priority, as it is a real stumbling block, especially for new users.

If the file format can't be determined automatically, we should explicitly ask the user in the next step!

uellue, 2018-10-11:
Full ack! :+:

sk1p, 2018-11-27:
> If the file format can't be determined automatically, we should explicitly ask the user in the next step!

Everything except this part is now implemented.

uellue, 2021-02-16:
Autodetection seems to work well for all formats we support by now. Closing here!

closed 2021-02-16 by uellue