LiberTEM/LiberTEM #67: Access Control: Authentication and Authorization

sk1p, 2018-08-15:
LiberTEM currently can be used to access all data of the user that it is running as, and different analysis methods may be used to exfiltrate confidential data. Using the file browser, the file system can be explored by any users that can access the web interface.

I would distinguish the following deployment scenarios that have different security requirements (from #36, updated):

1) Single-user system: LiberTEM can run as the login user and thus access all files of the user, and should only listen on the loopback interface. No need for authentication or authorization, as it would add complexity without a real threat. We may want to add a token that is shown on the command line when the server is started, which is needed to access LiberTEM. This is also defense-in-depth against DNS rebinding or related attacks.

2) Shared computer, listening on loopback: LiberTEM should run as a separate (non-privileged) user, and can only access users' files if they allow it via standard file system measures. As long as LiberTEM only listens on a loopback interface, no trust boundary is crossed: anyone who can locally access or SSH into the system could access those files directly anyways. This would not work if users need to access confidential data sets.

3) Shared computer, listening on public address: authentication (and SSL) becomes necessary. If any of the data is confidential, further access checks (authorization) become necessary, maybe including a mapping of "web user" to UNIX users. This can become messy. Imagine spawning I/O processes as the respective UNIX user which then somehow communicate with the worker processes. Easiest way: fd passing via UNIX domain socket. But that would not work for all file formats (case in point: HDF5 does not allow opening via fd). Or there could be a layer above, that handles authentication and spawns a whole LiberTEM server for each user, which of course costs more memory, but would delegate all file access control to the OS (see also: jupyterhub). We can't really support this in the first release, so the plan would be to remove (or disable) the ``--host`` parameter for now.

4) Cluster with HDFS: [Authentication and Authorization via Kerberos](https://hadoop.apache.org/docs/stable/hadoop-project-dist/hadoop-common/SecureMode.html) - users authenticate with their LDAP credentials, authorization via kerberos is integrated with HDFS. Ingestion should allow to set an umask with a secure default (077 to disallow all other users access by default). I think most work would be actually the initial HDFS setup and making sure the hdfs3 python lib works well with kerberos.

Integration into larger data management solutions should also be on our radar, like [Globus](https://globus.org) as mentioned by @davidelbert or the [EUDAT services](https://www.eudat.eu/services/). 

Checklist:

 - [ ] Remove ``--host`` for the first release (0.1)
 - [ ] token based authentication check

sk1p, 2018-12-08:
For token-based auth, we can use the stdlib package `secrets`, specifically [`secrets.token_urlsafe()`](https://docs.python.org/3/library/secrets.html#secrets.token_urlsafe) and [`secrets.compare_digest()`](https://docs.python.org/3/library/secrets.html#secrets.compare_digest).

uellue, 2023-10-25:
It seems that JupyterHub as a proxy to add SSL and authentication works just fine. Since this issue has been stale for so long, I'm closing for now.

closed 2023-10-25 by uellue