LiberTEM/LiberTEM #70: On-the-fly cropping of raw dataset could be faster

sk1p, 2018-08-18:
Cropping a `(256, 256, 130, 128)` raw dataset to `(256, 256, 128, 128)` on the fly is currently about three times slower than persisting the cropped version to a new raw dataset and working on that instead. This needs to be improved, especially once we support culling of frames to the bounding box of all masks, see #28 .

sk1p, 2018-08-18:
Instead of cropping, we can zero-pad the mask and should be back to full speed. Needs to be introduced in a general way (i.e. the dataset should be able to say "there is X bytes non-image data before/after each frame, please ignore", job/analysis implementations should then be able to decide whether they want the cropped data or uncropped). For #28, it may be necessary to find a trade-off between reading less data and the performance implications of cropping.

sk1p, 2018-08-20:
Just a note: cropping a numpy array creates a non-contiguous array, which numpy doesn't seem to handle well.

Zero-Padding worked well in a micro-benchmark, but didn't work out when used in LiberTEM. Don't know why yet. Maybe this is related to non-power-of-two dimensions?

In c89d1a885825400b468bdcabb24f427c66ed4e6d I added support for using [torch](https://pytorch.org) as a backend for matrix multiplication, which handles all cases (pre-cropped, non-contiguous array, zero-padding) really well, and may be a future avenue of implementing GPU acceleration. The improvement  is not because torch uses mkl, numpy+mkl was about as slow as numpy+openblas in my tests.

uellue, 2021-09-06:
Upon testing I didn't notice a difference between opening the same file as RAW uncropped and with the EMPAD reader, neither Linux nor Windows. Plus, the back-end code has changed completely since. Closing!

closed 2021-09-06 by uellue