LiberTEM/LiberTEM #71: Frame Average Preview times out

sk1p, 2018-08-20:
When loading large datasets, the frame average view can timeout after some minutes (on chrome: 2). This can be fixed by using the usual job API instead of the simple preview URL in an image tag. Using the job API also has the advantage of giving better feedback by showing partial results.

sk1p, 2018-09-13:
Checklist for implementation:

- [ ] augment reducer state by a reference to the job for average image / frame picking
- [ ] when creating an analysis, fire off the job for average image
- [ ] when switching from average to pick, stop the averaging job and start the new picking job
- [ ] update `<Preview/>` component to use `<ResultImage/>`
- [ ] on closing datasets or removing analyses, also cancel the frame picking and/or averaging jobs

After this is done, we should have better feedback, less resource waste, a more responsive way of working, ...

For example, on very large datasets, this means the user can switch over to pick mode, which stops the averaging job, create his mask(s), and have a good result even for the first pass over the whole dataset.

sk1p, 2018-10-11:
Done as of c489ecd8956b5d1609750aa1b87ccbaba351341b

closed 2018-10-11 by sk1p

uellue, 2018-10-11:
Tested on Windows 10: Preview image job shows progress, canceled upon closing. Nice!