LiberTEM/LiberTEM #72: HDF5 performance: try memory mapping

sk1p, 2018-08-21:
Continuous data from hdf5 files can be mmapped (see also: [notebook with prototype implementation](https://gist.github.com/maartenbreddels/09e1da79577151e5f7fec660c209f06e), [discussion in this ticket](https://github.com/BlueBrain/HighFive/issues/87))

 Interesting to know:

 - does it work at all? (it should)
 - how does it compare to chunked datasets?
   - processing speed (while reading from SSD and from FS cache)
   - write speed (could be mmapped as well, right? parallel writes without MPI)
   - file size: how much overhead does chunking have?

Of course for datasets like that we lose support for the more advanced HDF5 features, like compression.

AnandBaburajan, 2021-01-18:
> how does it compare to chunked datasets?
>   * processing speed (while reading from SSD and from FS cache)

@sk1p Hello! I've a doubt: does that mean I should compare the processing speed of the current LiberTEM master on a chunked dataset with the speed of my mmapped version on a continuous version of the same dataset? I'm not able to understand how that comparison would help. Please let me know if I got something wrong, thanks!

I'm sorry for staying out of touch for a while. I hope you're doing well!

sk1p, 2021-01-18:
> does that mean I should compare the processing speed of the current LiberTEM master on a chunked dataset with the speed of my mmapped version on a continuous version of the same dataset? 

The idea was to compare the same operation on different HDF5 files of the same data, saved with different data layouts (chunk or not, chunk size, maybe compression etc.) This is coming more from the perspective of giving recommendations to data producers. People sometimes ask us about our experience with different formats and possibilities of handling data, and one question is, if it even makes sense to produce chunked HDF5 files. It may turn out that, at least on SSDs and Linux, tiled reading from HDF5 via mmap could be faster than reading chunks via the HDF5 library. If the data is already in FS cache, using mmap should be even more favorable: we can basically read the data in a zero-copy fashion (at least if we don't have to convert dtype etc.) - with h5py and libhdf5, there is at least one copy done when reading chunked data, which can be painful on larger data sets.

So in other words, you don't need to worry about this, we can try the non-chunked/chunked/chunk-size/chunk cache size etc. parameters independently of your work. If you want to benchmark, maybe compare the mmaped reading of a non-chunked dataset with reading via h5py? I'm happy if it is as fast as before - very likely it will perform even better.

> I'm sorry for staying out of touch for a while. I hope you're doing well!

No worries, we all have lives :grin: Thank you for continuing with this work!

AnandBaburajan, 2021-01-19:
> This is coming more from the perspective of giving recommendations to data producers. People sometimes ask us about our experience with different formats and possibilities of handling data, and one question is, if it even makes sense to produce chunked HDF5 files.

I see. I get it now, thanks!