LiberTEM/LiberTEM #74: Controlled pre-fetching

sk1p, 2018-08-23:
Using `madvise`, one can tell the kernel that the memory in a region will be needed, using `MADV_WILLNEED`. Or for file descriptors, the same can be done with `fadvise64` and `POSIX_FADV_WILLNEED`.

The consequence is that the kernel will pre-fill the page cache for us, and even weird access patterns within the prefetched area will be fast, for example reading a `(1, 16, y, x)` slice if we manage to convince the kernel to prefetch 16 frames for us.

I guess on Windows this would be done using [`PrefetchVirtualMemory`](https://msdn.microsoft.com/en-us/library/windows/desktop/hh780543(v=vs.85).aspx).

uellue, 2020-04-09:
Refs #473

sk1p, 2020-04-09:
See also #706 : https://github.com/LiberTEM/LiberTEM/blob/08c93838dc5f546fcfe3b9916eeaff055ce9379e/src/libertem/io/dataset/base/backend.py#L225 :grinning: - this can be made more intelligent still, and needs to be benchmarked, but it's a start.

uellue, 2022-07-28:
On Linux the kernel seems to do a good job with prefetching. This would mostly be a topic for Windows to solve the performance issues with memory maps there.