LiberTEM/LiberTEM #75: COM Notebook: DaskJobExecutor: Failed to connect

woozey, 2018-08-23:
Trying to reproduces results in the  [COM notebook](https://github.com/LiberTEM/LiberTEM/blob/master/examples/center_of_mass.ipynb), got following:
```python
executor = DaskJobExecutor(scheduler_uri="tcp://localhost:8786", is_local=True)

---------------------------------------------------------------------------
TimeoutError                              Traceback (most recent call last)
/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/comm/core.py in connect(addr, timeout, deserialize, connection_args)
    185                                           future,
--> 186                                           quiet_exceptions=EnvironmentError)
    187         except EnvironmentError as e:

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/tornado/gen.py in run(self)
   1132                     try:
-> 1133                         value = future.result()
   1134                     except Exception:

TimeoutError: Timeout

During handling of the above exception, another exception occurred:

OSError                                   Traceback (most recent call last)
<ipython-input-15-4dcb2a5993da> in <module>()
----> 1 executor = DaskJobExecutor(scheduler_uri="tcp://localhost:8786", is_local=True)

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/libertem/executor/dask.py in __init__(self, scheduler_uri, client, is_local)
     13             if client:
     14                 raise ValueError("pass either client or scheduler_uri, not both")
---> 15             self.client = dd.Client(scheduler_uri, processes=False)
     16 
     17     def close(self):

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/client.py in __init__(self, address, loop, timeout, set_as_default, scheduler_file, security, asynchronous, name, heartbeat_interval, serializers, deserializers, extensions, **kwargs)
    614             ext(self)
    615 
--> 616         self.start(timeout=timeout)
    617 
    618         from distributed.recreate_exceptions import ReplayExceptionClient

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/client.py in start(self, **kwargs)
    737             self._started = self._start(**kwargs)
    738         else:
--> 739             sync(self.loop, self._start, **kwargs)
    740 
    741     def __await__(self):

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/utils.py in sync(loop, func, *args, **kwargs)
    275             e.wait(10)
    276     if error[0]:
--> 277         six.reraise(*error[0])
    278     else:
    279         return result[0]

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/six.py in reraise(tp, value, tb)
    691             if value.__traceback__ is not tb:
    692                 raise value.with_traceback(tb)
--> 693             raise value
    694         finally:
    695             value = None

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/utils.py in f()
    260             if timeout is not None:
    261                 future = gen.with_timeout(timedelta(seconds=timeout), future)
--> 262             result[0] = yield future
    263         except Exception as exc:
    264             error[0] = sys.exc_info()

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/tornado/gen.py in run(self)
   1131 
   1132                     try:
-> 1133                         value = future.result()
   1134                     except Exception:
   1135                         self.had_exception = True

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/tornado/gen.py in run(self)
   1139                     if exc_info is not None:
   1140                         try:
-> 1141                             yielded = self.gen.throw(*exc_info)
   1142                         finally:
   1143                             # Break up a reference to itself

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/client.py in _start(self, timeout, **kwargs)
    825         self.scheduler_comm = None
    826 
--> 827         yield self._ensure_connected(timeout=timeout)
    828 
    829         for pc in self._periodic_callbacks.values():

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/tornado/gen.py in run(self)
   1131 
   1132                     try:
-> 1133                         value = future.result()
   1134                     except Exception:
   1135                         self.had_exception = True

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/tornado/gen.py in run(self)
   1139                     if exc_info is not None:
   1140                         try:
-> 1141                             yielded = self.gen.throw(*exc_info)
   1142                         finally:
   1143                             # Break up a reference to itself

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/client.py in _ensure_connected(self, timeout)
    862         try:
    863             comm = yield connect(self.scheduler.address, timeout=timeout,
--> 864                                  connection_args=self.connection_args)
    865             if timeout is not None:
    866                 yield gen.with_timeout(timedelta(seconds=timeout),

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/tornado/gen.py in run(self)
   1131 
   1132                     try:
-> 1133                         value = future.result()
   1134                     except Exception:
   1135                         self.had_exception = True

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/tornado/gen.py in run(self)
   1139                     if exc_info is not None:
   1140                         try:
-> 1141                             yielded = self.gen.throw(*exc_info)
   1142                         finally:
   1143                             # Break up a reference to itself

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/comm/core.py in connect(addr, timeout, deserialize, connection_args)
    193                 _raise(error)
    194         except gen.TimeoutError:
--> 195             _raise(error)
    196         else:
    197             break

/usr/local/anaconda2/envs/libertem/lib/python3.6/site-packages/distributed/comm/core.py in _raise(error)
    176         msg = ("Timed out trying to connect to %r after %s s: %s"
    177                % (addr, timeout, error))
--> 178         raise IOError(msg)
    179 
    180     while True:

OSError: Timed out trying to connect to 'tcp://localhost:8786' after 10 s: in <distributed.comm.tcp.TCPConnector object at 0x7f78cc63f710>: ConnectionRefusedError: [Errno 111] Connection refused
```
I assume that the port might be simply closed. Or is it something more advanced?

sk1p, 2018-08-23:
So, this snippet of code may be a bit misleading:

```python
if True:
    cluster = dd.LocalCluster(threads_per_worker=1)
    client = dd.Client(cluster)
    executor = DaskJobExecutor(client=client, is_local=True)
else:
    executor = DaskJobExecutor(scheduler_uri="tcp://localhost:8786", is_local=True)
```

The first branch (that is always executed via `if True`) creates a new local dask cluster, the second connects to an existing cluster. So unless you have a cluster running, only execute the first part.

You can also just use the new `libertem.api` - it creates a local cluster for you.

woozey, 2018-08-23:
Ah, ok! I thought it is exactly opposite! Thanks!
I also have tried `InlineJobExecutor()` - it worked. What is it actually?

sk1p, 2018-08-23:
Here is a quick&dirty way to use the center-of-mass stuff as it is visualized in the web interface:

[image]

Let me know if this helps.

`InlineJobExecutor` is a non-parallel `JobExecutor` implementation - it just iterates over the partitions and processes them one after another.

sk1p, 2018-09-10:
The center of mass notebook was updated to use the new Python API in c7eee5c3b7846a9b83fb812e61ae9a76f3e22a88, so I'm closing this one.

closed 2018-09-10 by sk1p