LiberTEM/LiberTEM #80: Don't litter cwd with dask-worker-space

sk1p, 2018-08-26:
It is created by `dask.distributed` wherever you start LiberTEM - it should instead be created in a fixed location (`~/.cache`? Mac OS X / Win?)

uellue, 2018-08-27:
@sk1p what is in that folder? 

1. Should it persist between sessions? 
1. Might it contain temporary data of significant size that we should delete in case LiberTEM doesn't exit cleanly? 
1. Does it contain things like lock files that are relevant for concurrent sessions or that might mess things up if they linger in case LiberTEM didn't exit cleanly? 
1. Can two instances share the same folder?
1. Might data accumulate over time, like log files? 
1. Is the IO performnce or latency of the underlying file system performance-critical somehow?

uellue, 2018-08-27:
Did some research:

* We can have large temporary data here
* SIGKILL will make it linger
* The workers will try to get a lock for their directory

https://github.com/dask/distributed/pull/1543
https://stackoverflow.com/questions/48657374/what-is-the-default-directory-where-dask-workers-store-results-or-files
https://distributed.readthedocs.io/en/latest/worker.html#spill-data-to-disk
https://github.com/dask/distributed/issues/1540
https://github.com/dask/distributed/issues/2113

Is this a different directory?

https://github.com/dask/dask/issues/1659

Deactivate dumping data there. I guess we should do that because we read from files anyway and it should never be necessary, if I understand correctly. I'd rather have workers die than clutter disk space.

http://dask.pydata.org/en/latest/setup/hpc.html#no-local-storage

So that would mean: 

1. Should it persist between sessions? No.
1. Might it contain temporary data of significant size that we should delete in case LiberTEM doesn't exit cleanly? YES, but not for LiberTEM PROVIDED everything works as it should!
1. Does it contain things like lock files that are relevant for concurrent sessions or that might mess things up if they linger in case LiberTEM didn't exit cleanly? No, since workers only lock their own dir. But the file system needs to support locking.
1. Can two instances share the same folder? Yes, it seems, since each worker creates their own unique folder.
1. Might data accumulate over time, like log files? Leftover dirs can accumulate when the worker (or scheduler?) dies unexpectedly.
1. Is the IO performance or latency of the underlying file system performance-critical somehow? YES, if it is used for large intermediate data. Should not be relevant for LiberTEM IF everything works as expected.

What about using %TEMP% \libertem\dask on Windows? That expands to C:\Users\<user>\AppData\Local\Temp\libertem\dask on my system (Windows 10).

uellue, 2018-08-27:
On Unix this sounds like a job for a private subdirectory in `/tmp/`, I guess?

uellue, 2018-08-27:
Or maybe this is relevant? https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html

robbmcleod, 2018-09-28:
Probably what you want to use here is the `tempfile` module:

https://docs.python.org/3/library/tempfile.html#tempfile.TemporaryDirectory

sk1p, 2018-10-01:
Thanks @robbmcleod - I wasn't aware of the `TemporaryDirectory` class. It appears to be a nice, modern approach to `mkdtemp`! Temporary storage should also work nicely (always supports locking AFAIK), at least for our case of no spilling.

sk1p, 2018-10-30:
There is also https://pypi.org/project/appdirs/ - specifically `user_cache_dir`

samcmill, 2019-01-31:
+1

This is also an issue for read-only container images.

sk1p, 2019-05-27:
Closed via #345

closed 2019-05-27 by sk1p