LiberTEM/LiberTEM #81: Open browser after starting server

sk1p, 2018-08-26:
Call `webbrowser.open(...)` on startup - should be able to disable this. Also need to make sure to wait until after we are listening.

twentyse7en, 2020-02-06:
What are the prerequisite required to solve this issue ?

sk1p, 2020-02-06:
> What are the prerequisite required to solve this issue ?

Basically, you need to figure out how to integrate the `webbrowser.open` call into the [command line interface](https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/web/cli.py) at the right place, such that the web browser opens at the right time. You may also want to look at the [main and run](https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/web/server.py) functions of `libertem.server` - that's likely the right place for integration.

As there should be an option to disable this functionality (for example, because you are running the `libertem-server` on a headless system, that is, one without a GUI system running), there should also be a command line option to disable this behavior.

Let me know if this clears things up for you!

twentyse7en, 2020-02-09:
Explanation is pretty clear . So when we start server by `$libertem-server` it should open web browser by default. There should be also option to disable it.

uellue, 2020-02-21:
Closed with #580 

Many thanks @twentyse7en !

closed 2020-02-21 by uellue