LiberTEM/LiberTEM #83: Filebrowser improvements

sk1p, 2018-08-27:
- [ ] recent files / recent directories (could also remember the dataset parameters)
- [ ] HDFS browsing
- [ ] better convey status: disable file browser UI while loading directory listing (important for high latency situations)

uellue, 2018-08-27:
More suggestions:

- [ ] Separate "Up" icon, not just the ".." directory since that is unfamiliar to Windows users.
- [ ] "Back" and "Forward" icon
- [ ] Configurable shortcuts to common directories with user data
- [ ] Input window for path to type or copy&paste the path
- [ ] Display general metadata -- size, creation date, update date, owner, ...
- [ ] Display specific metadata, for example from known HDF5 dialects or sidecars
- [ ] Use the available screen space efficiently. Right now it uses only part of the screen.

By the way, we are probably not the first ones to implement a web-based file browser. Is there something we can use?

sk1p, 2018-09-06:
As mentioned by @woozey: for Windows, the user needs to be able to switch to different drives (C:, D: etc., maybe even network drives?). For portable drives, we may need a reload button (should have dual functionality: reload list of available drives, reload current directory listing)

sk1p, 2018-10-30:
See also: #141 - better error handling needed

uellue, 2018-11-28:
@sk1p makes sense. Size, ctime, atime and owner should be sufficient metadata for the listing.

sk1p, 2018-11-28:
More portable version for listing drives, that lists mount points on unix in psutil: https://psutil.readthedocs.io/en/latest/#psutil.disk_partitions - which we already depend on anyways

sk1p, 2018-11-28:
General metadata is now displayed, but styling can still be improved (especially for narrower screen widths)

uellue, 2018-11-28:
> General metadata is now displayed, but styling can still be improved (especially for narrower screen widths)

Awesome, love it! I really missed the size information before.

sk1p, 2018-12-04:
Input field is now interactive, still needs better error handling. Currently, on an error response when changing current directory, we bail out to '/', which is safe but not really convenient (btw. how does this work on Windows? is '/' equivalent to C:\?)

uellue, 2018-12-04:
@sk1p AFAIK Windows is not necessarily installed on C:\, and C:\doesn't always exist, although that is definitely the default. https://superuser.com/questions/655961/is-it-possible-for-a-windows-installation-to-not-have-the-c-drive

The correct way to determine the system drive is to look at the %windir% environment variable. That will be C:\WINDOWS on most systems, but will point you in the right direction when things are different.

Here's a list of all environment variables on Windows 10: https://docs.microsoft.com/en-us/windows/deployment/usmt/usmt-recognized-environment-variables

What about moving up the given path until a directory exists? Having to navigate all the way back from / into a deep nested directory structure because of a typo or some other mix-up would be annoying. 

If we want to default to home, here's how to determine that in a portable way: https://stackoverflow.com/questions/4028904/how-to-get-the-home-directory-in-python

sk1p, 2018-12-04:
> @sk1p AFAIK Windows is not necessarily installed on C:, and C:\doesn't always exist, although that is definitely the default. https://superuser.com/questions/655961/is-it-possible-for-a-windows-installation-to-not-have-the-c-drive

The question was more if '/' would be equivalent to the system drive, i.e. what happens if you do `os.listdir('/')` on a windows system.

> What about moving up the given path until a directory exists? Having to navigate all the way back from / into a deep nested directory structure because of a typo or some other mix-up would be annoying.

Yes, that was my idea too. Just need to be careful: when the top level is reached, it still may fail, because the drive letter is now unavailable, or you may no longer have access to that directory. In that case, we should list drives again and pick another one. Or picking the home directory may be a good idea in this case, as you are guaranteed to have access there.

Let's think about this: Windows is installed on Z:\, another user inserted his USB stick, mounted on E:\. Let's also say we have a USB stick inserted as F:\. We are browsing F:\Data\Big\. We now remove our USB stick, and open the Browser again. It fails to list F:\Data\Big\ and proceeds to try to go up, until F:\, which also fails. If we pick the 'next' drive, this may also fail as E:\ is not accessible by us.

> If we want to default to home, here's how to determine that in a portable way: https://stackoverflow.com/questions/4028904/how-to-get-the-home-directory-in-python

Yep, already using that for the "Go to... → Home" functionality.

sk1p, 2018-12-05:
> What about moving up the given path until a directory exists? Having to navigate all the way back from / into a deep nested directory structure because of a typo or some other mix-up would be annoying.

This is now implemented with a fall back to `Path.home()` in case the top level is not reachable. If you have time, could you please test this on Windows? Maybe one case where the drive letter doesn't even exist, and one where some base of the path exists. The latter should be tested already via the test cases, but I'm not sure how to test the former.

About using the screen space efficiently: we need to keep it responsive with regard to screen height. One possibility would be to just make the file browser full-screen, but that means you lose the context: what datasets are already open? Did I want to open X or Y? Without full screen, you can just scroll a bit and you have your context.

Another possibility is to just make the height of the file browser a percentage (80%?) of the viewport height, with a minimum height, so if you resize your browser window, it stays usable.

The "configurable shortcuts" section - I think this may actually fit under the "Go To..." -> "Places" button; there could be "add shortcut" or something, and a remove thingy for existing shortcuts.

I'm not so sure how to do the metadata displaying - only: the user should be able to open it from the list of files, and it should also be visible together with the open form.

uellue, 2019-09-24:
See also https://github.com/LiberTEM/LiberTEM/issues/186#issuecomment-534518019

uellue, 2023-10-25:
Closing since the majority of boxes are checked and the remaining ones haven't been touched for four years.

closed 2023-10-25 by uellue