LiberTEM/LiberTEM #84: Simple API: Check if file exists, and perhaps sanity check, upon Context.load()

uellue, 2018-08-28:
Right now, [`Context.load()`](https://github.com/LiberTEM/LiberTEM/blob/a465da9ab48c7224b376869be318c7794d011a98/src/libertem/api.py#L17) accepts a file that doesn't exist. An error occurs only in [`Context.run()`](https://github.com/LiberTEM/LiberTEM/blob/a465da9ab48c7224b376869be318c7794d011a98/src/libertem/api.py#L66)

Would it make sense to perform a consistency check upon `load()` so that the process fails at the earliest possible time? Things to check for:

* File exists and can be opened
* Type matches for formats like K2, HDF5 etc
* HDF5 with internal path: That internal path does exist and matches the expected type
* Size matches the given layout

sk1p, 2018-08-28:
The validity check method of the dataset is now called when loading. The checks per `DataSet` need to be improved though. Some can be interesting, for example `Size matches the given layout` should not be enforced as an error, maybe it should become a warning: until we have built-in support for cropping datasets, the scan size can be used for controlling how much to read for some datasets (raw, k2is at least). Also, for some files, there may be garbage at the end which you don't want to read (k2is allows for partial data blocks / frames at the end of the files)

uellue, 2018-08-29:
Ok, thx, works for me now! :-) I guess we can close this issue and open separate ones for specific checks as needed.

closed 2018-08-29 by uellue