LiberTEM/LiberTEM #88: Shorthand to fetch a complete frame at scan position (x, y)

uellue, 2018-08-28:
Right now, one has to use this method to pick a frame: https://github.com/LiberTEM/LiberTEM/issues/63#issuecomment-415475726

This method imports details from the internal API and can select whole slabs of data for processing.

However, when writing a script one often just wants to show one complete frame, for example as a reference. The method referenced above is unnecessarily complicated for that task. The Python API should have a shorthand function to load one complete frame at scan coordinates (x, y).

uellue, 2018-09-04:
Currently, the code to fetch a frame looks like this:

```python
get_sample_frame = PickFrameJob(dataset=ds, slice_=Slice(origin=(60, 75, 0, 0), shape=(1, 1, detector_y, detector_x)))
res = ctx.run(get_sample_frame)  # res is a numpy array with shape as specified above
sample_frame = res[0,0]
```

This call can be simplified in the following ways:
* The full 4D slice is not necessary. Only the X, Y coordinates in scan dimension need to be specified.
* The result is always one frame, that means rs can immediately be a 2D numpy array.

The code could look like this:

```python
get_sample_frame = PickSingleFrameJob(dataset=ds, origin=(60, 75))
sample_frame = ctx.run(get_sample_frame)
```

uellue, 2018-09-04:
Another point that was a bit confusing for me was the order of X and Y. I normally expect X to be in first place, while the first index in LiberTEM results is the Y index. Well, as long as we keep it consistent and documented, that should be fine. :-)

sk1p, 2018-09-04:
How about the following:

```python
job = ctx.pick_frame(dataset=ds, y=60, x=75)
sample_frame = ctx.run(job)
```

- the user doesn't need to import any job implementation, just the `Context`
- having separate parameters for x and y means the user can pass them as keyword arguments and doesn't have to care about the order: `ctx.pick_frame(dataset=ds, x=75, y=60)` works as well

uellue, 2018-09-04:
@sk1p Perfect! :-) 💯

uellue, 2018-09-04:
Another one: What about having an optional "offset" parameter to identify a frame by its offset instead of X and Y? Could be useful for vector scans or the K2 format debugging "fun" #63

commit cb47af9e (2018-09-04): Python API: create_pick_job(dataset, x, y), fixes #88
closed 2018-09-04 by sk1p