LiberTEM/LiberTEM #106: Optimize handling of result tiles

uellue, 2018-09-04:
Discussion with @sk1p :
Since LiberTEM workers calculate results very quickly and process the data in relatively small portions for optimal cache efficiency, handling the incoming result data stream can put heavy load on the master node. This can quickly become a limiting factor for installations with many nodes.

Suggestion by @sk1p : Accumulate data at the workers per partition / task instead of feeding back small tiles.

See also #105 !

sk1p, 2018-11-14:
Work towards this is done in d968c9b46e77472c04857c35b4ead41227d4f388 and d6179772a8130c4417b71cb4f386e51d7ea8057e. Still need to benchmark though.

sk1p, 2018-11-14:
Results on 18 core machine:

Times are best-of-N (there are other processes running on the machine that I don't have control over...), I just used disk analysis with default parameters (so applying one mask). `tileshape` was 1,8,128,128 for the raw dataset.

|before/after | ~25GB raw float32 | ~8.1GB K2 |
|-----------|---------------|------------|
|before |   2.5s | 1.45s |
|after |  1.55s | 0.9s |

So for both cases, it takes about 0.6x as much time!

Delegating tiling to the BLAS library gives another ~100ms improvement for the raw dataset, but this is not consistent (maybe because of other running processes, or because of AVX512 slowdown?)

sk1p, 2018-11-14:
Ugh, okay, I was stupid when running those benchmarks, and forgot to set the cpufreq governor - we need some kind of reproducible environment for benchmarking! Here is an updated table with governor=performance, other parameters as above. New column for delegating tiling to BLAS:

|before/after | ~25GB raw float32 (delegate tiling) |  ~25GB raw float32 | ~8.1GB K2 |
|-----------|-----|----------|------------|
|before | 1.2s | 1.8s | 1.11s |
|after |  1.2s | 1.43s | 0.71s |

Now delegating tiling to BLAS gives a consistent speedup, and as the changes are only affecting tiling, there is no change before/after in the 'delegating' column.

uellue, 2018-11-14:
Do I read correctly, shy of 21 GB/s? 😳 Congratulations!

uellue, 2018-11-19:
@sk1p I guess we can close this issue now that it works and gives great performance? :-)

closed 2018-11-19 by uellue