LiberTEM/LiberTEM #108: K2IS: efficient uint12 decoding

sk1p, 2018-09-05:
See #63 for previous discussion, https://github.com/LiberTEM/LiberTEM/tree/master/prototypes/k2/uint12-decode for initial analysis and reference implementations.

sk1p, 2018-09-06:
I pushed an initial implementation in d04f866187c48fee768ec3a7a10b30c03c8eb205 and the following commits. In a micro benchmark, on an old i5-2410M, it is about 3 times faster compared to the numba jit version (on a Xeon(R) W-2195, the difference is not as pronounced, factor ~1.8).

For vectorization, eliminating the output buffer was necessary, writing directly to the output works well.

sk1p, 2018-09-06:
Using AVX512 (`-march=skylake-avx512 -mtune=skylake-avx512`) on the Xeon W2195 makes quite a difference (only tested on a single threaded workload for now): it is then again ~2.8 times faster than the numba jit version. This is what the compiler generates: https://godbolt.org/z/Elv9ZJ

sk1p, 2018-09-06:
As an interesting fact, the memcpy call is optimized to 4 `vmovups` instructions!

sk1p, 2018-09-06:
So just with something like:

```
 __attribute__((target_clones("avx2","arch=atom","avx512", "avx2", "default")))
void uint12_decode(...)
```

We can make gcc (version >=6) build multiple versions of our decoding routines that are dynamically selected based on CPU capabilities. Sadly, it doesn't work as-is with clang, but the basic building blocks should be there. Multi-versioning should also only really be considered for the distributed binary builds, and are not really useful for day-to-day development.

Travis CI does supports a current clang/gcc [with something like this](https://github.com/dreal/dreal4/commit/898497a72fd4f24ff05e9398fb170d9650fe315d), so we could get away with requiring a specific compiler version for the multi versioned build.

uellue, 2018-09-06:
@sk1p :+1: great! How possible would it be to use numba for the math? I've tried it before with my experiments, and numba went back to Python because of type casts so that it was slow. However, I haven't tried [view()](https://docs.scipy.org/doc/numpy/reference/generated/numpy.ndarray.view.html) yet. That should do only one typecast per buffer.

What is the best way to try that out? Technically, the Python reference implementation in the Jupyter notebook should be quite possible to put in place of the current numba version?

That would save us a LOT of trouble with development and deployment!

sk1p, 2018-09-06:
I tried to implement with numba, but didn't manage to improve on the naive version yet, that just loops over the input and processes 3 bytes at a time. A quick comparison:

[image]

First one is the numba implementation based on the jupyter notebooks of @uellue, second one is the naive version, third one is the C++ implementation.

sk1p, 2018-09-07:
I got [the "naive" version of the decoder](https://github.com/LiberTEM/LiberTEM/blob/74d5d81e1de118eb69437be350fc075a04366561/prototypes/k2/uint12-decode/impl/u12decode.cpp#L37) to auto-vectorize by partially unrolling the loop (to be more specific: by helping the compiler to unroll the loop). Annotating input and output with `__restrict__` is also necessary for vectorization. In my tests it was as fast as the other C++ implementation or faster. More next week.

uellue, 2018-09-07:
Perfect! The less code, the better, the faster, the better. :-)

I'm a bit curious why the numba version is slower. 
* Is it because it doesn't have the unrolling and equivalent of `__restrict__` for the compiler to vectorize?
* Is it compiling for a target without SIMD?
* Is the source code or intermediate code more complicated for the compiler so that the optimizer can't do as much work as in pure C/C++?
* Is this a result of more advanced language-specific optimizers for C++ compared to what numba can use?
* Is the resulting code doing more complicated things because of being derived from Python?

I'm just asking because being able to produce highly optimized, vectorized machine code from Python with numba would reduce the development effort @sk1p  laid out in https://github.com/LiberTEM/LiberTEM/issues/108#issuecomment-419073198

In that sense it would be AWESOME if a numba version could reach the performance of a C/C++ extension.

sk1p, 2018-09-07:
Hmm, good questions. To be clear, we are talking about the "reference" numba implementation that is currently used by the k2is reader, right?

> Is it compiling for a target without SIMD?

It is definitely compiling for an AVX2 target, it is using AVX instructions for some data movement. I'll try to help the compiler unroll the loop on the numba version, too.

> Is it because it doesn't have the unrolling and equivalent of __restrict__ for the compiler to vectorize?

I tried to find if `__restrict__` is necessary with numba, or if there is some syntax for that but didn't find anything yet.

sk1p, 2018-09-11:
The numba guys pointed me in the direction of https://github.com/numba/numba/issues/3205 - "Numba is possibly generating code that handles the wrap-around negative indexing that is unnecessary because the index expressions are all non-negative. We've observed this in a few situations, and are trying to come up with a fix."

sk1p, 2018-09-19:
See also: https://stackoverflow.com/questions/44735756/python-reading-12-bit-binary-files/45070947#45070947

sk1p, 2018-09-24:
The version by @max9111 (I hope I'm tagging the right user :grin:) is vectorized by numba and almost as fast as the C++ version (15 µs vs 18 µs)! I think the crucial difference to our numba impl is that the loop is incremented by 1 instead of 3. 

The encoding is a bit different - the left part of the middle byte belongs to the left uint12 vs the right uint12. I'll try to reformulate our decoder using increments of 1 in the loop and see if it gets vectorized or if the encoding plays a role here.

max9111, 2018-09-24:
Did you also try a parallelized version [nb.njit(parallel=True) and changing the loop to nb.prange]? 

This breaks SIMD-vectorization but is about 20% faster on a 4C/8T. On a 18C/32T machine this could be much faster than the single threaded SIMD-vectorized version.

A multithreaded and SIMD-vectorized version may be possible by using Pythons multithreading module (Numba releases the GIL) or some compiler hints which force vectorization.

sk1p, 2018-09-24:
Thinking about it some more: I measured the parallel version using `%timeit` in a jupyter notebook, and on a quite small buffer (as I'm doing the same in the real workload), so maybe there was some overhead starting up new threads for each call, which could be amortized by working on larger buffers. Also, I'm still using numba 0.39, and I heard the new version has quite some improvements in that area. Nevertheless, I'm quite happy with the results now, as the single-threaded version also fits well into our overall architecture.

uellue, 2018-09-25:
That is awesome, I'm impressed. I think we'll rely more on Numba in the future. :-)

uellue, 2018-10-04:
The [current Numba version of the decoder](https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/io/dataset/k2is.py#L30) has all the properties we want:

* High throughput, enabled through vectorization by the compiler
* Simple, understandable, compact code
* Avoids a C build chain for development or installation

It reaches the same order of magnitude in throughput as the fastest dedicated bit packing and unpacking implementations like https://github.com/lemire/simdcomp and https://github.com/powturbo/TurboPFor that go through a lot more trouble and use hand-optimized architecture-specific intrinsics. 

I'm closing this issue for now since @sk1p and I are satisfied with the current version and right now can't think of a better way to do this. :-) If you do have a better idea, let us know!

closed 2018-10-04 by uellue