LiberTEM/LiberTEM #115: Make implemented Analyses available via the Python API

sk1p, 2018-09-10:
The users of our Python API should not need to reinvent things that are already implemented In LiberTEM.  

I see two ways, at least:  a single method for all analyses, or one method for each analysis.

```
job = ctx.create_analysis_job("com", cx=64, cy=64, mask_r=24, rot=π). # better name?
# ....or...
job = ctx.create_com_job(cx=64, cy=64, mask_r=24, rot=π)
```

I prefer the first. It keeps the API concise and general.

The second does he the advantage of being friendlier for API usage, because it can have a specific docstring. I have an idea how to fix this for the first style, though... Which is also relevant for `Context.load()`: on import, a short docstring could be fetched for each available dataset, or analysis, and interpolated into the `load`/`create_analysie_job` docstring. Not as nice as the real deal, but better than nothing.

uellue, 2018-09-10:
A very good point, exposing this capability with the API! :-)

Honestly, in an intuitive way I'd prefer functions with different names. Maybe with a common "job" namespace, such as "ctx.jobs.com.create()", if you want to avoid clutter? That looks a lot like Java, btw...

From my intuition it feels "wrong" to pass sth that constitutes a class name as a string parameter and an name=argument list to a "**" function, then use the class name to switch to a delegate function that gets the parameters dumped on it. IMO the class should be declared as a class, and used as one. Or is this the preferred way to do things in Python? Maybe my past with Delphi, C and Java is shining through? :-D

sk1p, 2018-09-10:
Well, we are not passing in the class name, just some kind of identifier for it :grin: (imagine it as an enum). This is actually an implementation of the [factory pattern](https://en.wikipedia.org/wiki/Factory_method_pattern) - see the C# example there. The positive side of this is that, when adding a new analysis type, we don't need to change the API - it is automatically included after putting it into the "registry" once (see [the `load` implementation for an exmaple](https://github.com/LiberTEM/LiberTEM/blob/master/src/libertem/io/dataset/__init__.py). For analysis we won't need to dynamically import like for loading, as we shouldn't have optional dependencies there). The negative side, well... bad doc string? Also, may be a little hard to follow the code, with this layer of indirection.

What exactly did you mean by  'the class should be declared as class'? Should the user manually import and instantiate a `libertem.analysis.disk.DiskMaskAnalysis` class? It is currently built for the web API, so it just accepts a dict of parameters in the constructor, but that can be changed.

uellue, 2018-09-10:
Well, I meant class in a more abstract way, whatever thing that tells you what object it is and which implementation of a common functionality to call. Do we have a way to autogenerate API docs from everything that's in the dictionary?

To be honest, I didn't fully understand why changing the API is bad. I mean, it is indeed changing if we change which types work and what parameters we accept and require. The only point is that we hide it from the API documentation and from the user of that API. Maybe I am missing sth, but making the API more obscure that way is not exactly an advantage, or is it? The only advantage I can imagine is when your pay bonus depends on having a stable API but you want to change the API all the time. That way you can do that without losing your bonus. Or when your users want weak typing, but use a strongly typed language that would have to be recompiled if the function changes.

uellue, 2018-09-10:
What I meant was, when the language and framework are _statically_ typed, but you want _dynamic_ typing.

Hmm, I was looking at the factory pattern in a bit more detail. I think it makes sense in cases where objects are created in the same way with the same parameters, you just need to switch the type. In that case the docstring doesn't matter because it would be the same for all functions, and you actually streamline your documentation.

In cases like our job creation where the parameters vary wildly and the resulting objects do very different things and just have in common that they implement a very basic common interface, I'd say that separate functions are cleaner and more transparent for the user. I think it is never a question "Do I want a `PickFrame` job or rather a `MaskJob` to do this task". It is always very clear which one to use.

With file loading it is a different story IMO. The function does pretty much the same thing, and it makes sense to be able to not specify a type at all and let the load function find out which type to use. In that case the factory pattern feels right to me. :-)

Maybe this could be a good decision point if the factory pattern makes sense: Is it thinkable to not specify the type at all and let an internal logic figure out which type to instantiate?

In any case, in the end both options work as long as we keep it consistent. How do other Python APIs and frameworks handle this, by the way?

sk1p, 2018-09-13:
Matplotlib is [an example](https://matplotlib.org/api/_as_gen/matplotlib.pyplot.axes.html#matplotlib.pyplot.axes) of how not to do it: here is a bunch of `kwargs`, and by the way, the arguments for function a and b are also allowed. According to [a discussion on the python subreddit](https://www.reddit.com/r/Python/comments/9fb9i3/am_i_the_only_one_who_hates_matplotlib/), this is one of the most-hated "features" of the mpl API. Also, nice things like this: "In fact, completely unknown kwargs are often accepted and then just ignored[...]!"

So, that is definitely not what we want. On the other hand, I would dislike to repeat full method signatures. It's not about not changing the API, it's about changing the API at exactly one place. We will end up with diverging APIs. But having thought about it more, this is a very good thing: it provides isolation between the user-facing Python API and our HTTP API or other internal APIs, which are currently not documented and as such should be considered unstable.

So, what can happen:

 * Analysis gets an additional parameter (example: center of mass now properly supports rotation). The Python API is not updated. Either, the parameter is optional and it just isn't supported until someone notices (boo), or the parameter is required by the analysis, thus would be a breaking API change, which can be caught by testing (which means we should make it an optional parameter if possible!)

* Analysis parameter is removed. Testing catches this and hopefully the removal is re-thought in some backward-compatible way (example: proper deprecation warning is raised, and it is removed in a future version).

* Analysis parameter changes in some way. May be caught in testing, or maybe not. Not different to the proxying/factory-like way of things.

So, in conclusion, I will create full methods for each analysis, and proper testing is paramount for keeping the API stable! I'd like to make it a goal to have at least 80% testing coverage (unit testing and integration testing combined) for 0.2 - and complete testing coverage for the user-facing API. Do you agree that this should be a focus for 0.2?

uellue, 2018-11-20:
Bumped priority because this is useful for @FWin22.

sk1p, 2018-11-21:
Center of mass, disk, ring, point analyses are now available via the API.

commit 5fd7feb1 (2018-11-21): Make implemented analyses available via the API

 + Fixes #115
 + This change also removes the out parameter of Context.run, which makes
   little sense of the output is more post-processed, like it is with
   fully-fledged analyses like COM, where we return separate channels etc.
 + Also extend the Python API usage docs with some result handling
closed 2018-11-21 by sk1p