LiberTEM/LiberTEM #120: Deal with binned data

sk1p, 2018-09-11:
Data can be binned unevenly on the x and y dimensions, which means we basically have non-square pixels. For a 4-times binning on y direction, our pixels would be 4px tall and 1px wide. As an example, the PNdetector hardware supports increased frame rates (up to 20k I think?) by binning in the y direction, similar to the K2.

- support for specifying pre-binning as input parameter(s) (which formats? RAW, FRMS6, HDF5, k2is?)
- in processing - if you want to apply a circular mask, you either need to expand the binned data, or fix your mask to have the same pixel size (either by binning the mask the same way the input data was binned, or by applying the according geometric transformation when drawing the mask)
  - pro expanding data: any subsequent processing should "just work"™
  - con expanding data: we have much more data to deal with! if people really do record data with 20k FPS by binning 20 times, we want to still be able to process it in (soft) real time!
   - I could imagine that some processing algorithms are not as easily adapted to non-square pixels, so
maybe this is actually something that needs to be decided per-job

Idea: there should a possibility to `map` a function (or multiple?) over all tiles as a stream transformation, which can then implement the expansion in a general fashion (no need to build support for each dataset etc.). Note that this will have direct impact on the selection of the input tileshape (the _expanded_ form of each tile has to fit into cache etc.). The job implementation may need some kind of capability function that says: "I can process non-square pixels", or "I can only process full frames" etc.

uellue, 2018-09-11:
Hmm, that opens a much larger topic about rendering the data, sparse sampling with non-rectangular, non-uniform scanning, non-standard detectors and the likes. I'd be strongly against blowing up data in basic processing. We should rather transform the input mask into the correct shape for the detector and render the output data and sample frames into the correct shape for on-screen rendering.

The transformation can be expressed as a (sparse) matrix-matrix multiplication, if I recall correctly, matching input channels into the corresponding output channels. [PyFAI](https://github.com/silx-kit/pyFAI) is doing sth like that, matching detector pixels with high precision and antialiasing into radial bins.

uellue, 2018-09-11:
Hehe, it is time to rekindle the work on NeXus. There we have a framework to express all this in the file format.

uellue, 2018-09-11:
By the way: https://pytorch.org/docs/stable/sparse.html

With that we can do high-throughput binning as well, by the way.

uellue, 2018-09-11:
Also good for scaling, if I may add.

uellue, 2018-09-11:
When the matching is periodic, one can probably use a "matching template" for one periodic region and stamp that over the entire thing. Wrapping around the matrix indices or sth like that. Not unlike the bit unpacking.

sk1p, 2018-09-12:
I'm not sure I understand what you mean by "arbitrary detector layouts" - can you elaborate and maybe give a few examples?

> Hmm, that opens a much larger topic about rendering the data, sparse sampling with non-rectangular, non-uniform scanning, non-standard detectors and the likes

Why do we have to solve non-rectangular scanning together with binning? I don't really see the connection. Sparse sampling, arbitrary scan positions etc. would mostly concern the shape of the result, no? I mean, I see how having for example a smaller scanning step size in one direction would create a result image scaled in that dimension, but beyond that, there is so much more flexibility needed in the scanning dimensions than in the pixel format. Can't we separate "layout of the pixels" from "layout of the scan"? Are there even sensors used for microscopy that have a weird pixel layout beyond those "simulated" by binning? Things like SuperCCD with hexagonally arranged pixels?

> I'd be strongly against blowing up data in basic processing. We should rather transform the input mask into the correct shape for the detector and render the output data and sample frames into the correct shape for on-screen rendering.

Yes, for our built-in analysis this is true, and quite easily done, and I totally agree that we should. I was more thinking about ease of development: if all our analyses _have_ to support binned input data, it would put a higher learning curve on new developers. If we had a conversion to square pixels in place, we could have techniques implemented and in LiberTEM progressively: start with something that works and is correct, and then scale it up to optimal performance. I mean, we are following this pattern elsewhere too: we [know that we can read the k2 format faster](https://github.com/LiberTEM/LiberTEM/issues/108), we [know that we can speed up small masks by cropping](https://github.com/LiberTEM/LiberTEM/issues/28), but we can _use it now_, because we allowed for imperfect code.

You'll have to explain the sparse/matching idea in person, I don't really follow right now.

> it is time to rekindle the work on NeXus

Agreed!

uellue, 2018-09-12:
@sk1p "Arbitrary detector layout" means that people have a lot of ideas on putting sensitive areas (formerly known as pixels) in different places:
* Detectors with a central hole (Rafal wants that to combine with EELS)
* Detectors with some pixelated areas, some with concentric rings (Colin Ophus)
* Detectors with elongated pixels (effectively what PNDetector needs)
* Tiled detectors with, perhaps, small gaps between the tiles (Medipix 4? What about Robert's chip?)
* Data binned for ptychography -- I've seen it binned in all kinds of shapes to reduce the calculation effort

uellue, 2018-09-12:
> Why do we have to solve non-rectangular scanning together with binning?

First of all, for the calculation of standard pixelated STEM virtual detectors we flatten the scan and detector dimension so that we get a 2D matrix. That means in terms of number crunching, the scanning pattern and detector layout don't matter. We just have to know how to produce the mask vectors and display the result after.

The whole thing both for rendering vector scans and for handling weird detector layouts boils down to matching pixels or channels to real-world coordinates, the way I understand it. That means for rendering a vector scan, for rendering a mask onto a detector layout and for rendering a detector frame in the GUI we can use the same internal facilities. 

Actually, thinking about it, we have three cases: 
* Render "weird" discrete channels (formerly known as pixels) from and to square pixels, for example for on-screen display of detector data or for transforming square pixel masks to mask vectors for a detector.
* Render from vector shapes, for example to create the mask vector from primitives like circles etc.
* Render vector scans where we get a list of (x, y, time) as (float, float, float) coordinates. 

We just have to solve this once properly -- which I think is not too difficult -- and we support EVERYTHING I can imagine right now.

uellue, 2018-09-12:
> If we had a conversion to square pixels in place

You mean we offer to first feed the data through a rendering algorithm to make it square, and then give it to the processing? I like the idea, especially the aspect of blowing it up only for small chunks at a time, processing, and discarding! What about Karina's algorithm? That might already be a case where we can use it. I have to talk to her when I am back.

The same facility can be used for all kinds of binning jobs if we want to process reduced data. If we do this well, we can always do it on the fly with relatively little overhead. 

Thinking even further, this is a way to implement algorithms that want a Fourier transformed images or any other transform as input, for example. We could have a small list of highly optimized filters so that people developing some processing routine don't have to worry about that. We still need to see if it makes more sense to keep the Fourier transformed around rather than re-calculating each time, though.

uellue, 2018-09-12:
> start with something that works and is correct

Good point. Maybe a pragmatic approach would be to first implement the rectangular pixels with simple scaling, but already thinking what the general architecture would look like, and implement the scaling in a way that fits the bigger picture?