LiberTEM/LiberTEM #121: Be able to work in physical units rather than pixel

uellue, 2018-09-12:
At the moment, LiberTEM only sees pixels. In the future we should work with real physical units for display and calculations.

sk1p, 2019-01-14:
Maybe something we can learn from astropy: http://docs.astropy.org/en/stable/units/

uellue, 2019-07-18:
Related project to solve this in a general fashion: https://github.com/LiberTEM/nexus-4dstem

uellue, 2019-07-18:
LiberTEM UDFs should allow the user to choose between receiving data with detector corrections (gain, dark image, dead pixels) or performing corrections in the UDF since some applications, namely dot product with masks, allow to correct the masks instead of the data.

Generally, the LiberTEM facilities should focus on generating corrected masks where that is possible since that is much. much better performance-wise in most cases.