LiberTEM/LiberTEM #122: Handle live acquisition with a simple GUI

uellue, 2018-09-12:
There's a lot of demand to handle live data streams from the detector. We should implement at least a basic graphical user interface (and scripting capability) to hook up a detector to LiberTEM, stream the data to file and allow the user to explore the data while it comes in and accumulates.

uellue, 2018-09-12:
Yes, Merlin should be the first candidate because we have already @magnunor 's code and the interface seems benign.

uellue, 2018-09-14:
Just a "shower thought", what about using a video stream to show the updating data? 

* Should compress well -- not much change per frame, probably much more efficient than image fragments for high refresh rate
* Should be working natively in modern browsers these days
* Optimized for silky-smooth display with low resource use on the client -- uses GPU and works smoothly on any smartphone

What about resource use for encoding? In any case, it shouldn't be worse than sending JPEG snippets. Which codec would work? MP4, WebM llok good: https://developer.mozilla.org/en-US/docs/Web/HTML/Supported_media_formats#Browser_compatibility

sk1p, 2018-09-14:
> - Should compress well -- not much change per frame, probably much more efficient than image fragments for high refresh rate
>  - Should be working natively in modern browsers these days
>  - Optimized [...]

Agreed!

> shouldn't be worse than sending JPEG snippets

Don't know about that. Do you have a comparison handy? Basically comparing CPU usage for encoding h264/h265/... vs ~MJPEG. As far as I know, "real-time" h264 encoding is only possible either because of [hardware support](https://wiki.archlinux.org/index.php/Hardware_video_acceleration#Formats) or good multi-core usage, where we would rather use these cores for more useful work.

The [MJPEG Wikipedia page](https://en.wikipedia.org/wiki/Motion_JPEG#Encoding) just hand-waves that it "is not computationally intensive", I'd say citation needed?

A comparison I found:

http://blog.loopbio.com/video-io-2-jpeg-decoding.html

They are comparing single-core ~MJPEG with multi-core (how many?) h264 etc. - so if that benchmark was run on a 6 core machine, I would guess that a trivial parallel implementation of MJPEG could beat h264. Oh well...

Either way, before going into the performance side of video encoding, maybe we should talk about the requirements?

 - latency?
 - frame rate?
 - image quality?
 - compression ratio / low bandwidth?
 - resolution / down sampling?
 - metadata synchronization?
 - are the requirements different between raw frames vs. analysis result?

And more generally: what workflows are needed? 

PS.: https://pypi.org/project/PyTurboJPEG/ and the fork by the loopbio folks at https://github.com/loopbio/PyTurbojpeg looks interesting, and it's quite compact! Also: they have an eye for [unnecessary copies](https://github.com/loopbio/PyTurboJPEG/blob/4e0bf23a3ac31fa38c2d5bd66e5aeeafd195eb4e/turbojpeg.py#L282), yay! :rocket:

sk1p, 2021-06-10:
I think we can close this issue, as jupyter notebooks work quite well for simple GUI needs. What do you think?

closed 2021-06-10 by uellue