LiberTEM/LiberTEM #124: Implement fast ptychography

uellue, 2018-09-12:
Lots of people are interested in ptychography right now. We should add that capability to LiberTEM. Since it is our strength, we should aim towards a fast, parallel, distributed implementation. Here we can discuss the details.

woozey, 2018-09-12:
As far as I understand ptychography in TEM is still not a fully developed method. It still faces quite critics from people with strong theoretical and practical background. It would still require numerous methodological developments and comparison of results with other phase techniques. On the other hand reconstruction algorithms can be made available with a disclaimer that result quality assessment and interpretation is up to users...

uellue, 2021-02-16:
Closing here, see https://github.com/Ptychography-4-0/ptychography!

closed 2021-02-16 by uellue