LiberTEM/LiberTEM #126: Calculating continues even if a dataset is closed

uellue, 2018-09-12:
I opened a dataset that was too big for my system to handle. I hit "Apply" and realized after some waiting time that I needed to stop the calculation. So I closed the analysis and the dataset.

What I expect to happen: The calculation stops immediately and my system resources are freed.

What actually happened: The GUI closed the dataset, but the calculation ran in the background until the end. In the log I could see that result tiles kept coming in.

sk1p, 2018-09-12:
This is partially fixed by 668df58. The next step is to fix #71 - we are currently simply loading the preview and picked images via an image tag in the web interface, which also means there are no sane cancelling semantics in place. This should be purely a client-side change, other than removing the preview/picking handlers from the web API. This also has the nice effect of localizing the dependency on `dask.distributed` to only the executor.

uellue, 2018-09-26:
Ok, seems to work now. It looks like LiberTEM stops the calculation a few seconds after the dataset is closed. There are no more result tiles coming in after closing and system load drops to idle state almost immediately. I'm keeping it open because @sk1p is working on cancelling the preview image as well in #71.

sk1p, 2018-10-11:
#71 is fixed, and so should this issue!

closed 2018-10-11 by sk1p

uellue, 2018-10-11:
Tested on Windows 10, confirmed to work!