LiberTEM/LiberTEM #129: Reader for MIB

uellue, 2018-09-26:
With this Issue we document the state of the MIB (Quantum Detector Merlin) reader. 

- [ ] More testing with different files and eliminating all issues.
- [ ] Support for binned data.
- [ ] Performance optimization #53.

sk1p, 2018-10-15:
We also need to think about coordinate systems: MIB frames have their origin at the bottom left, we have it at top-left:

[image]

sk1p, 2018-11-13:
Note: there is still a bug with border handling, needs more testing!

sk1p, 2018-11-21:
Frame pick performance improved as per #170. Sending jobs involving all partitions has degraded "a bit" though. I looked into it and saw that lots of time is spent pickling. I think this is because the dataset object is sent over for all tasks, and so is the generated cache of partitioning data. We may be able to filter that out of the pickled data.

A possible way around this would be to build a "JobContext" object that contains objects that stay the same for all tasks. This object can then also contain other constants like the masks.

I need to look more into it but I think this can then be "sent into the cluster" using `scatter` functions.

sk1p, 2018-11-21:
> Note: there is still a bug with border handling, needs more testing!

In the meantime, to work around this, be sure to chose a tileshape that evenly divides a scan row (If you don't care about performance, chose something like `(1, 1, 256, 256)`).

sk1p, 2018-11-21:
Also, there is a bug if the number of frames in the MIB files don't match up with whole scan rows, even though I haven't hit it yet. This will be fixed in the tileshape refactor.

uellue, 2019-06-18:
We didn't come across binned data yet. In general the MIB reader has proven itself in practice. I am closing this and we fix issues as they arise.

closed 2019-06-18 by uellue