LiberTEM/LiberTEM #130: Improved data-task affinity for cluster

uellue, 2018-09-26:
Currently, LiberTEM in connection with Hadoop FS checks which cluster nodes hold which blocks in their local storage and assigns tasks to these nodes.

Since Hadoop FS replicates the data, subsequent jobs might assign the tasks to different nodes that hold the same data, which requires reading it again and displaces other data from the file system cache.

In the future, LiberTEM should assign tasks for the same blocks preferentially to the same nodes as in previous runs because they hold the data in their file system cache, which results in much better use of the collective RAM of the cluster.

This might require augmenting the [dask.distributed scheduler](http://distributed.dask.org/en/latest/locality.html?#user-control) since it currently supports a mandatory set of equal nodes, or a soft scheduling to any nodes with some preferred ones.  The improvement described in this issue would require a more nuanced figure of merit for each node.

uellue, 2018-10-02:
As a side node, this should work particularly well for clusters that are based on a shared POSIX network file system like NFS or Lustre because those don't have any natural data-task affinity. This enhancement should dramatically improve FS cache efficiency for that case.

As a side question, is the mapping already deterministic somehow, i.e. provide the same preferred mapping if a job is re-run, or does that vary? TODO check with new cluster.

sk1p, 2019-02-06:
I did a very simple implementation for task affinity, where tasks for the same partition always gets executed on the same host. This depends on the partitions always being returned in the same order.

uellue, 2019-11-08:
Ok, I guess we can close this for now? The caching layer should probably take care of that, too? :-)

closed 2019-11-08 by uellue