LiberTEM/LiberTEM #132: Averaged preview image depends on partition shape with .blo file

uellue, 2018-10-02:
I was experimenting with the "Tileshape" parameter while loading the Glasgow BaTiO3 / BiFeO3 .blo data set.

Observed behavior: The averaged preview image on the left depends on the first two items of the "Tileshape" setting.

Expected behavior: The averaged preview image is the same, independent of tileshape.

Examples:
(90,121,144,144)

[image]

(1,8,144,144)

[image]

(8,121,144,144)

[image]

(8,70,144,144)

[image]

(8,8,144,144)

[image]

(90,8,144,144)

[image]

(90,121,12,12) -- same as (90,121,144,144)

[image]

uellue, 2018-10-02:
Looking at the data set, it seems that only an average of the first partition is taken. That explains why the very prominent (110) reflections  of the central region are only present in images with a large partition size. With small partitions, only an average of a border region is taken that doesn't contain the prominent (110) reflections.

sk1p, 2018-10-04:
This should now be fixed in master, but leaving this still open as I still need to write a test for it.

uellue, 2018-10-04:
@sk1p Thank you! Works for me now. :-)

sk1p, 2018-10-04:
Hm, the test works for me locally but fails in CI: https://travis-ci.org/LiberTEM/LiberTEM/jobs/437113763 - need to investigate.

sk1p, 2018-10-04:
Should be fixed, was missing a commit when cherry picking.

commit dde282c0 (2018-10-04): add test for 19c8876, closes #132
commit 87edbfb2 (2018-10-09): add test for 19c8876, closes #132
closed 2018-10-04 by sk1p