LiberTEM/LiberTEM #134: GUI rework/rethink

sk1p, 2018-10-04:
There are several drawbacks to the way the current GUI works:

 * only works well for square-ish scan regions and square-ish frames. tall or very wide scans are scaled down without a way to zoom in
 * does not work well with high-resolution images
 * does not use screen real estate well
 * not flexible enough for complex workflows

For 0.2, I'd like to refactor the UI to make the following possible:

 * zoom into/out of both picked frames and results
 * fullscreen view of images
 * more complex workflows, like: make HAADF image, then pick a representing frame from the HAADF image, then compute something based on the selected frame (example: distance of all frames from selected frame according to some metric)

The following could be of interest:

 * http://golden-layout.com/tutorials/getting-started-react.html - resizable/reorderable tabs/split-views/windows (see https://godbolt.org/ for a live example)

When we have decided what kind of workflows we want to support in 0.2 and in the future, this issue should be updated accordingly.

uellue, 2018-11-20:
We should have a clean concept on how to show context information for an analysis. Right now the info under frame and result window is a bit unstructured, just concatenating various text bits and elements, and adding a new item therefore needs tinkering. This made moving the pick coordinate display #164 more difficult than necessary, for example. I was wrestling with separator characters and `&emsp;` ...

sk1p, 2019-01-08:
There are several related issues in the [GUI project](https://github.com/LiberTEM/LiberTEM/projects/2), #235 restating the problems with large frames.

sk1p, 2019-01-08:
For visualization changes like colormap selection (#166), value range, log scale etc (#128). we will have the same problem as in #235 - image encoding will limit the performance and hence the interactive response time. We may want to rethink the visualization aspect, and maybe move it to the client.

For colormapped visualization in 2D: https://github.com/santilland/plotty https://av.tib.eu/media/20373

For possible 3D visualization: http://deck.gl/#/examples/custom-layers/3d-surface-explorer

For a more fully featured framework: https://gr-framework.org/tutorials/using_gr_in_javascript.html

Deck.gl and plotty.js rely on WebGL (plotty actually has a fallback to plain old canvas) - we have to think if our users always have it available. On the other hand, WebGL is a natural fit for the problem, and allows to offload visualization to the GPU. Implementing a zoomable interface should also work well with WebGL. Deck.gl already implements a zoomable interface by integrating with mapbox-gl-js, but that's not really a good fit for our application, I think. I mean, one could create a 'projection' that fits the physical units, and position the raster data on that "map"...

uellue, 2019-01-08:
https://gr-framework.org/about.html

> GR was developed by the Scientific IT-Systems group at the Peter Grünberg Institute at Forschungszentrum Jülich. The main development has been done by Josef Heinen who currently maintains the software.

Nice! If there are no strong reasons against it, I'd favor that. :-)

uellue, 2019-01-09:
Discussions with @sk1p: Consolidated requirements for GUI display.

# Flexible, interactive display

* Zooming
* Panning
* Restrict operations to ROI?
* Changes in display style
    * Color map, clipping, log scale for 2D data #128, #166 
    * Curve, range, log scale for 1D data
* Control with mouse
* Control with keyboard

# Units

* Display all axes in physical units with label, scale or scale bar, ticks, color scale bar, etc #121

# Interactive controls

* Selectors for virtual detectors and pick controls
* Selectors for integration range in curves, including background reference(s) (element maps)
* Selectors for ROI?
* Control with mouse
* Control with keyboard
* Highlight focus #211
* See also #212
* Allow several controls per image?
* Implement background substraction selector for diffraction disks (Glasgow example)?
* Generally, make it easy to implement custom controls for custom analysis

# Smoothness

* Smooth, interactive feeling with real-time response (100 ms target) to user input
    * Handle suitable operations like changing range or color map completely on the client
* 60 fps target for live data streams.
    * Test if frame dropping or reduced constant rate are better if conditions don't allow for 60 fps

# Performance

* Only transfer data that is required for a current view since frames can be large
    * Reduced data for zoomed-out or cropped views
* Rate of change in the data buffer provided by LiberTEM
    * Live beam monitor: 60 fps 4k frames
    * Calculation result: about 200 updates per second (8 node cluster, 14 GB/s processing, K2IS data proportions with partition shape (750, 1860, 256)); size 500 kB (sum image) or 750 B (virtual detector). This figure can be reduced with pre-reduction #242
* Rate of change in GUI: Target 1920 x 1080 @ 60 fps, minimum 256x256 @ 30 fps (?)

# Quality of service

* Beam monitor: Dropping frames or reduced frame rate is OK. Low latency is desirable.
* Update of scan image: Include all partial updates to avoid holes in the image.
* Adaptive rate?
* Parallel operation of several display "slots" with different characteristics within the GUI without interference
    * Beam monitor and updating scan image
    * Several parallel data channels
    * Mix live acquisition and offline processing in one GUI
    * Probably instantiate separate connection and back-end for each slot to allow the operating system and network stack to balance between the separate slots.

# Data types

* Pixelated detector or 2D scan
    * X and Y pixel size can be different
* Spectral (curve)
* Raster multi-channel (complex numbers, result of PCA or strain analysis, for example) 
    * Assign colors to channels and mix -- transparent overlay etc?
    * Select and deselect channels?
    * Mathematical operations to combine channels?
* Raster with discrete key/class per pixel (result of clustering or thresholding)
    * Color per key
    * Display only selected keys
* 2D vector (center of mass)
[... 26 lines cut ...]
* Support re-using client-side display module with different back-end implementation and different web GUI framework.

# Ideas for implementation

* LiberTEM provides data buffer with current result for display
    * LiberTEM is responsible to assemble the data
    * LiberTEM signals changes to the Web API server, including changed bounding box
    * Web API server signals changes to LiberTEM that require LiberTEM to re-calculate the result, for example changed GUI controls
* Web API server extracts data required for the current view
    * Receives commands from renderer and potentially GUI controls
    * Filters data based on current view and sends data to renderer through transport
    * Propagates changes originating from LiberTEM to the renderer
* Renderer displays results
    * As far as possible, reacts to changes of controls directly, for example a changed color map
    * Asks web API for updated data as far as required, for example during zooming and panning
    * Updates display based on new data received from web API server
* The transport encodes the result.
    * Inner layer: Application-specific format, for example data types
    * Outer layer: Encapsulation, for example http requests or web sockets

The stack of web API server, transport, renderer and UI controls is instantiated for each display slot to make sure that they can work independent of each other with a minimum of interference. Critical example: High data rate with large frames from live monitoring should not inhibit other GUI updates, for example scan result or calculation result.

# TODO

Are there existing implementations of a system with similar requirements?

sk1p, 2019-01-22:
As part of the display system, it should be possible to have multiple raster data sets displayed with affine transformations applied to each. Would be nice for combined display of multiple data sets from the same sample, if we can put them into the same global coordinate system.

uellue, 2019-05-06:
Perhaps matplotlib's webagg https://matplotlib.org/gallery/user_interfaces/embedding_webagg_sgskip.html could be relevant for this?

uellue, 2019-11-28:
@unsimulatable

uellue, 2019-12-10:
Additional requirement: Allow handling structured result sets. The [Radial Fourier Series](https://libertem.github.io/LiberTEM/app/amorphous.html#radial-fourier-series) would be a prime example of an Analysis with a potentially very large and complex result.

The GUI should allow to display and navigate such a result set with ease and efficiency.

sk1p, 2020-10-12:
For inspiration: https://hacks.mozilla.org/2019/04/pyodide-bringing-the-scientific-python-stack-to-the-browser/ - for running Python-based visualizations on the client (for example matplotlib)

sk1p, 2021-03-17:
To hit the performance numbers, one could use zstd compression on the transport and something like https://github.com/AlanRace/zstd-wasm on the client side