LiberTEM/LiberTEM #136: Cluster ingest without Hadoop FS

uellue, 2018-10-08:
Discussion with @sk1p 

In the context of #105, having Hadoop FS as a requirement to use local SSD storage can make deployment complicated. In order to benefit from local storage without Hadoop FS in a cluster that has some form of shared FS, we could use the local storage as a form of cache.

Essentially, we set up the number of workers for our kubernetes cluster. When we load a data set, we split it into logical partitions and assign these partitions to specific containers. We map processing tasks only to the containers that have its partition assigned.

When receiving a task, the worker in the container then checks if it has the block in local cache. If not, it first ingests it from the source file in the shared file system and stores it locally, just like our Hadoop FS storage would do.

Since the cache is backed by a (hopefully reliable) shared file system, we don't need redundancy for the blocks. If a kubernetes container fails for some reason, we just fire up a new one, which will then regenerate its cache as needed. 

If we change the number of containers for our cluster, we should track which partitions should be deleted from the local cache of which container. That means we somehow unassign a partition and tell the worker to forget it? Will this actually happen in operation or will people start a whole new cluster?

We should generally manage the cache somehow , make sure we stay in sync with the source file and remove unused or stale copies. How "expensive" is local cache and do we share the total free space with others?

Is there code out there that can manage such a cache?

uellue, 2018-10-11:
More discussions with @sk1p 

First, we have to decide if LiberTEM manages the cache or if the operating system does that with [FS-Cache](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/6/html/storage_administration_guide/ch-fscache) or similar.

We came to the conclusion that LiberTEM should allow _both_. If we have a NAS mounted via NFS, FS-Cache is easy to set up and requires no change to LiberTEM to cache read access to the source file. This gives us a (hopefully) robust and tested solution that can handle sharing the cache between users and managing the storage space. For Lustre, a [similar solution](http://wiki.lustre.org/images/0/04/LUG2018-Lustre_Persistent_Client_Cache-Xi.pdf) is under development. [IBM Spectrum Scale](https://www.intel.com/content/dam/www/public/us/en/documents/white-papers/performance-gains-ibm-spectrum-scale.pdf) seems to offer such functionality as well. A system like [dCache](https://www.dcache.org/) could work as well.

Since we want to cache selected intermediate results, for example a Fourier transform for @kruzaeva's work, @sk1p and I think that we also need a cache managed by LiberTEM. This can also be used to cache objects from some form of cloud storage. This cache should be user-specific because LiberTEM shouldn't handle access control to such objects. If a shared cache is desired, this should be implemented on the cloud storage level independent of LiberTEM, for example with a HTTP proxy.

Question: How to manage inactive user's cache space? An old object in the cache would stay there because other users can't evict it. Proposed solution: Leave that to regular administration that has to look after user data anyway. Perhaps mount with atime and clean up automatically with a cron job if this becomes a burden?

Each data source should get a flag if LiberTEM or some underlying system handles caching of the raw data so that we don't keep double copies. One could by default not cache data in the file system tree and cache data from cloud storage.

To manage the space, we can start with a LRU strategy. A more advanced figure of merit could be taking into account the storage size vs. the reconstruction cost, i.e. transfer for input data or CPU for intermediate data.

We should limit cache space per user if we have many users on a system.

The required size for a cache object should be known in advance to make sure there's enough space -- evict other objects before, for example.

On the application level, one could implement the cache by passing result buffers to functions with cachable results. If caching is desired, the buffer is backed by a memory-mapped file. For further processing the output buffer is passed to the next step as input. Again, the input can be backed by a memory-mapped file with the cached data instead of re-calculating it. This is transparent both to the source function and the sink function.

If for some reason two independent processes create the same object in the cache at the same time, this doesn't conflict. The output data should be the same for the same input data. Both terminate as soon as they have completed their calculation. After that, the cache is guaranteed to contain a complete, correct result, independent of the second parallel process. As a drawback, the calculation is performed twice. However, having no locking eliminates a number of error possibilities, for example one job being terminated before finishing. Here, the second job would simply continue and complete the result without ill effect.

Caching should probably only be enabled for selected transformations, such as a Fourier transform. Candidates for caching are re-used often with the same input parameters, expensive to calculate, small result. This could include decompression if an expensive compression algorithm is used.

sk1p, 2018-10-11:
> We should limit cache space per user if we have many users on a system.

For LiberTEM-managed caching, this can be done by normal file system quotas.

> If for some reason two independent processes create the same object in the cache at the same time, this doesn't conflict. The output data should be the same for the same input data. Both terminate [...]

Passing the result buffer directly backed by the result file to the processing function may not be the best idea. For example, if you _aggregate_ into the result buffer, two processes working on the same file _will_ collide. I'd suggest working on a temporary file in the same file system and moving into place on completion. That means our cache will have the nice property of only storing completed and "correct" results, even in case of system failures (as moving files is atomic). These temporary files could be cleaned up on startup (NB: they can't be nameless to be auto-deleted on failure, I think, because you can't create names to arbitrary inodes). No locking needed still, as long as we don't do any fine-grained calculations that could cause excessive re-calculation (i.e. thundering herd).

sk1p, 2018-11-20:
See also: https://github.com/dask/distributed/issues/2294

uellue, 2025-03-26:
Closing since it is better to do this at the OS level.

closed 2025-03-27 by uellue