LiberTEM/LiberTEM #139: COM analysis yields a blank result when the ROI doesn't include the frame center.

uellue, 2018-10-18:
I was trying a COM analysis on individual diffraction spots of the Glasgow data set.

What happened:
I get a blank, black result frame when the ROI doesn't include the frame center. The bug described in #138 seems to be triggered when the border of the ROI is going through the frame center. It works when the frame center is inside the ROI.

What I'd expect to happen:
COM analysis works independent of the ROI within the frame.

sk1p, 2018-10-23:
Did you get blank results for all channels? Or only magnitude?

uellue, 2018-10-23:
> Did you get blank results for all channels? Or only magnitude?

Yes, all channels. But it only happens on the Glasgow data set. The EMPAD BiFeO3 works. 

Parameter | Value
-- | --
path | C:\Users\weber\ownCloud\Projects\Open Pixelated STEM framework\Data\3rd-Party Datasets\Glasgow\10 um 110.blo
type | BLO
name | Glasgow: Ian MacLaren, Shane McCartan
tileshape | (1,8,144,144)
shape | (90,121,144,144)

I confirmed that it is really the center of the frame that matters. If I shift the ROI only slightly to include or not include the center, the display changes dramatically from colorful to all-black after re-calculating.

PS: Awesome, I can copy & paste the "Info" table of the dataset and it keeps the table formatting!

sk1p, 2018-10-24:
Why this is happening: the dataset has areas (the very bottom left) that are completely free of intensity outside of the transmitted beam, which results in a divide by zero -> both min and max for normalization are `inf`.

uellue, 2018-10-24:
Hm, that makes sense. What would be he most correct behavior for this case? Can we somehow just ignore `inf` values for the color scale? Should we have a specific display for `inf` or `nan` pixels, such as white, black, grey, ...?

sk1p, 2018-10-24:
For center-of-mass, I would set the value to the reference center, which results in a zero-length vector and thus a neutral representation in the field visualization (and a zero in magnitude). I think this is the expected results in the case of center of mass with all weights equal to zero.

For the general case, we should discuss this in #128 I think.

closed 2018-10-24 by sk1p

uellue, 2018-10-24:
Confirmed fixed, thank you! :-) 👍

commit 25762f97 (2018-10-24): com: guard against divide by zero, fixes #139

img_sum can be zero at positions where the selected region of interest contains no
intensity. We now initialize those positions to the reference center
such that the resulting center of mass is zero there.

Note: numpy ufuncs, when using where= without specifying out=, return
uninitialzed memory at the places not selected by the where parameter.
This can be prevented by either passing in initialized memory with out=
or by explicitly initializing these indices. We do the latter as we can
save initialization costs for the whole result array that way [... cut]