LiberTEM/LiberTEM #140: Exception during processing makes the GUI wait forever

uellue, 2018-10-18:
When bug #138 is triggered, the GUI shows an eternal spinner. The waiting can be terminated by re-adjusting the ROI so that #138 is not triggered and clicking "Apply".

What I'd expect to happen:
If a job causes an exception and terminates prematurely, the GUI shows a helpful error message.

sk1p, 2018-10-23:
This is now fixed. For the asynchronous job processing we are doing, we explicitly need to send an error message over the websocket in these cases, and appropriately handle those in the GUI, which is now implemented.

closed 2018-10-23 by sk1p

uellue, 2018-10-23:
Great! I can't test it because I don't know an exception that I can trigger with the current version. I guess that's a good sign! :-D